openwrt/luci #788: IPv6 address validator does not understand mask

mindless2112, 2016-09-02:
LUCI does not allow the user to enter an address with an IPv6-address-style mask.  For IPv6 firewall rules, it is possible to use an IPv6 address in a form like `::0123:4567:89ab:cdef/::ffff:ffff:ffff:ffff` in order to match against the last 64 bits of the address, as you would expect.  (This is useful for setting up IPv6 firewall rules when your ISP provides a dynamic IPv6 prefix.)

cshoredaniel, 2016-12-10:
@Flameeyes @mindless2112 does this work with openwrt firewall program, or does it too need modifications? That is if you manually edit /etc/config/firewall as mentioned above, does it work?

jow-, 2016-12-10:
fw3 understands both ipaddr/bits and ipaddr/mask notations.

cshoredaniel, 2016-12-10:
@jow- thnx.