openwrt/luci #797: Unable to delete luci rule from device using luci UI

Deepak72840, 2016-09-14:
Hi All,

I am able to add the Delete button in UI, modified "rules.htm" file to add "Delete Rule" button.
I have added controller for delete rule in the file
package/luci/modules/luci-mod-admin-full/luasrc/controller/admin/boostingruledel.lua
package/luci/modules/luci-mod-admin-full/luasrc/view/admin_network/boostingruledel.htm

The delete function is getting called like "uci:delete("config","boost_rule",1)" from "boostingruledel.lua" file. However this does not deleting the rule from device.
 I am using the below command to delete the applied rule from device
# uci delete config.@boost_rule[0]
I am facing issue in passing the 3rd argument in delete function from
**"boostingruledel.lua" (uci:delete("config","boost_rule",1))**

Please can someone help me, in pointing how to pass the 3rd argument
and properly delete the rule ????
Thanks in advance

cshoredaniel, 2017-01-17:
@jow- Where does the LuCI API etc docs live these days?

cshoredaniel, 2017-01-22:
@Deepak72840 The third parameter is the name of the option you are trying to delete e.g for uci file:

/etc/config/config
containg:

`config boost_root 'boost_rule'
        option some_option 1`

it's be uci:delete("config", "boost_rule", "some_option")

HOWEVER that is a pretty weird thing to do from within LuCI because the CBI map should take care of it.  Can you clarify what it is you are trying to do, and show some sample source code and the goal?

cshoredaniel, 2017-01-22:
Also have you looked the README which points to http://htmlpreview.github.io/?http://raw.githubusercontent.com/openwrt/luci/master/documentation/api/index.html

jow-, 2018-07-03:
```
local x = require "luci.model.uci".cusor()

x:foreach("boost_root", "boost_rule", function(section)
    x:delete("boost_root", section[".name"])
    return false
end)
```

closed 2018-07-03 by jow-