openwrt/luci #803: Allowed connect ports description “A single value 0”

yydcool, 2016-09-22:
Based on the man page:
ConnectPort
This option can be used to specify the ports allowed for the CONNECT method. If no ConnectPort line is found, then all ports are allowed. To disable CONNECT altogether, include a single ConnectPort line with a value of 0.

a single ConnectPort line with a value of 0 is to disable all port.
The description is wrong:"List of allowed ports for the CONNECT method. A single value "0" allows all ports"

hnyman, 2016-09-22:
You are referring to this GUI item, right?

https://github.com/openwrt/luci/blob/master/applications/luci-app-tinyproxy/luasrc/model/cbi/tinyproxy.lua#L133
```
o = s:taboption("filter", DynamicList, "ConnectPort", translate("Allowed connect ports"),
translate("List of allowed ports for the CONNECT method. A single value \"0\" allows all ports"))
```

yydcool, 2016-09-22:
yes

hnyman, 2016-09-22:
You are probably right, it is defined that way also in the tinxyproxy package's config file here:

https://github.com/openwrt/packages/blob/master/net/tinyproxy/files/tinyproxy.config#L184
```
# This is a list of ports allowed by tinyproxy when the CONNECT method
# is used.  To disable the CONNECT method altogether, set the value to 0.
# If no ConnectPort line is found, all ports are allowed (which is not
# very secure.)
```

hnyman, 2016-09-26:
cc to tinyproxy package maintainer: @jow- 

Any view on this? Just a typo in Luci? (or alternatively in tinyproxy default config?) Or is there something in the uci handling that changes the behaviour?

M95D, 2022-05-01:
Can we fix this, please? It's just a typo in the description.

closed 2022-05-03 by jow-

M95D, 2022-05-04:
That was fast. Thanks!

commit eadf07b5 (2022-05-03): luci-app-tinyproxy: fix description of ConnectPort option

Fixes: #803
commit 71c51d31 (2023-04-03): luci-app-tinyproxy: fix description of ConnectPort option

Fixes: #803

(backported from commit eadf07b5299a450c43bc4162f891d5c95b629e8a)