openwrt/luci #806: luci-ssl - if not possible build with libustream-openssl (only libustream-polarssl) - way ???

build000, 2016-09-28:
In view of recent developments in the LEDE program Opkg, as well as scripts that build the image currently is not possible to build the image of both libraries at the same time (at least marked from the vending machine at the same time, when uhttpd choose openssl as the main support for uhttpd-mod-tls) - ie. libustream-openssl and libustream-polarssl, as it has done so far.
https://bugs.lede-project.org/index.php?do=details&task_id=189
I ask realize that practically ideologically it is pointless (it is the same name of the target library for both libustream-openssl and libustream-polarssl) but why users do not have to choose whether they want to use openssl or polarssl as a major libraries to support encryption of calls?
In short, it is impossible to practically use openssl to Lucia, you do not decide to polarssl what the railways blocking the entire mass of other programs that work only on the basis of openssl - this state of affairs is unacceptable and storm the idea of ​​free software.

jow-, 2016-09-28:
`luci-ssl` is an empty convenience meta package, don't select it if you do not want its default dependencies but pick `uhttpd-mod-tls`, `px5g` and your desired `libustream-*` flavor manually.

closed 2016-09-28 by jow-

hnyman, 2016-09-28:
Various packages support different encryption libraries. There is no single SSL library that supports all possible apps. Some of the apps have upstream support for just one library, while some have several alternatives. Developers have so far selected to use polarssl as the main default due to its small size. Some apps require openssl, and then that library gets included too.

px5g only supports polarssl & mbedtls, which forces one of them into the build. Feel free to craft a patch for px5g to add support for openssl, if you want that variant. 

It is possible to build luci-ssl functionality with libustream-openssl by using menuconfig. luci-ssl itself is just a "meta package" that selects luci, libustream-polarssl and px5g but contains no code. See https://github.com/openwrt/luci/blob/master/collections/luci-ssl/Makefile

Instead of selecting luci-ssl from the menu, do that manually and select luci, px5g and libustream-openssl. (px5g will still select libpolarssl library itself, but that does not hurt libustream.openssl.)

When selecting uhttpd-mod-tls, you are offered the selection of polarssl/mbedssl/openssl, so you can avoid libustream-polarssl also there.

Ps.
openssl is so large that it will never be the global default. Polarssl (and mbedtls, its newer incarnation) and cyalssl/wolfssl are so much smaller that one them will be the default due to size constraints in the smaller routers.
``` 
libpolarssl_1.3.17-1_mips_24kc.ipk                 28-Sep-2016 02:53              144310
libcyassl_3.9.6-1_mips_24kc.ipk                    28-Sep-2016 03:13              149797
libmbedtls_2.3.0-2_mips_24kc.ipk                   28-Sep-2016 02:53              162013
libopenssl_1.0.2j-1_mips_24kc.ipk                  28-Sep-2016 02:55              715251
```

jow-, 2016-09-28:
You don't strictly need px5g either if you have other means to create or ship certificates for uhttpd. You could eg. substitute it with a script that calls the openssl util.

hnyman, 2016-09-28:
> You don't strictly need px5g either if you have other means to create or ship certificates for uhttpd. You could eg. substitute it with a script that calls the openssl util.

@jow- 
That is actually a very nice idea.  I might look into it.

px5g-openssl variant that scraps the main px5g code and uses a script to create the certificate using openssl tools might actually work. That could enable openssl-only builds.

I used to do openssl-only build (with px5g-standalone that embedded parts of polarssl). But as px5g-standalone produces only SHA1 certs, it is not really usable any more, so I switched to normal px5g  a while ago.

build000, 2016-09-28:
@hnyman:

> When selecting uhttpd-mod-tls, you are offered the selection of polarssl/mbedssl/openssl, so you can avoid libustream-polarssl also there.

Except that it is currently impossible to build an image with both libraries at the same time (ie libustream-openssl and libustream-polarssl) in the image - https://bugs.lede-project.org/index.php?do=details&task_id=189

hnyman, 2016-09-28:
Please don't mix things. Nothing in SSL for LuCI requires you to have specifically libustream-polarssl.  Neither does px5g cause libustream-polarssl to be installed.  px5g's need for the underlying polarssl library has no impact in libustream.

Other packages, like cshark, are a separate issue. Even if LuCI has full support for openssl-only, you will still have problems if you try to use packages that explicitly require a different variant of libustream. Nothing done to Luci will fix that.

build000, 2016-09-28:
However, if LUCIE had to directly support only for openssl, whether in the form of some trap, with a choice globally, openssl or polarssl or even mbdtls - it would greatly facilitate the build version of the image at each other.
As for the issues that other trap may use different ustream than openssl is disproportionately something other than a conviction for native support polarssl in the LuCI - which disqualifies the ability to use many other interesting programs at the same time that the word polarssl but just blocks indirectly by libustream- openssl - and it is now a fact, after recent changes in opkg and building scripts.
It was thus denied the opportunity to easily and freely choose whether additional applications and global encryption method by the appropriate library is provided.
As for the argument that little it takes place (polarssl and additives) in the system but it is not decisive in every case - hence the need for actual choices at the "make menuconfig" from which you want to use and how to arrange the picture for our own needs without sacrificing at the same time with ease and freedom of use of LuCI.

hnyman, 2017-01-31:
@build000 
Both Openwrt and LEDE uhttpd have now been patched to support key generation with OpenSSL, so I committed https://github.com/openwrt/luci/commit/b6341bd197aed9640c4d0dcff73019e045978df0 to remove the dependency for px5g (and mbedtls/polarssl) from luci-ssl-openssl, which only selects OpenSSL tools from now on.

build000, 2017-01-31:
@hnyman 
I noticed - thanks for the info - the way I am using this form of Makefile from approx. one month and basically I do not see major problems in addition to obtaining the indicator "A" (https://www.ssllabs.com/ssltest/) ssl certificates indirectly produced by ACME, and therefore directly by the openssl-util (unfortunately compatibility with openssl md4 in LEDE is treated as a reduction of the safety certificate).