openwrt/luci #814: @hnyman: luci-ssl-openssl mistake by "placing image" - can not be compiled into firmware package luci-ssl-openssl (px5g selected as a module)

build000, 2016-10-04:
@hnyman
make -j1 V=s
```
(...)
Configuring usbreset.
Configuring luci-i18n-base-en.
Configuring kmod-fs-cifs.
Configuring luci-i18n-firewall-pl.
Configuring ncdu.
Configuring picocom.
Configuring comgt-hso.
Configuring luci-i18n-ntpc-pl.
Configuring uhttpd-mod-tls.
Configuring wireless-tools.
Configuring kmod-crypto-sha1.
Configuring zip.
Configuring acme.
Configuring ppp-mod-pppoe.
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-ssl-openssl:
 * 	px5g * 
 * opkg_install_cmd: Cannot install package luci-ssl-openssl.
package/Makefile:56: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/sj/Pulpit/Projekty/WNDR4700-LEDE'
package/Makefile:125: recipe for target '/home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/stamp/.package_install' failed
make[1]: *** [/home/sj/Pulpit/Projekty/WNDR4700-LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/sj/Pulpit/Projekty/WNDR4700-LEDE'
/home/sj/Pulpit/Projekty/WNDR4700-LEDE/include/toplevel.mk:193: polecenia dla obiektu 'world' nie powiodły się
make: *** [world] Błąd 2
```
... Because when px5g requires polarssl, and when we Strem fixed with uhhttpd-mod-tls for openssl a conflict - after the recent changes in opkg is unacceptable to install at the same time libustream-openssl and libustream-polarssl .... I mean I suppose that is causing the problem.
The ideal solution would be if apka luci-ssl-openssl required dependencies openssl-util, and some "automagically" library/program/script/pack type luci-app or luci-lib/etc., which included that in a script default to generate an SSL certificate on the basis of openssl-util - not as it is now solved that only px5g fully this role on the basis of cooperation with LuCI.

hnyman, 2016-10-04:
I am working on a full "OpenSSL-only" version. But the solution will need modification of the uhttpd init script on the LEDE & Openwrt side, too. (I have already patched my own build and it works now without px5g.) See my patches and discussion at https://github.com/openwrt/luci/pull/807#issuecomment-251226934

Like I said in our earlier discussion, px5g has no connection to any of libustream- packages and uses only libpolarssl (or possibly libmbedtls in LEDE), which causes no conflict. Once again: you can have several SSL libraries installed, but not several libustream- packages.

You edited version of this issue contains the correct reason for the error, I guess: px5g vs. px5g-polarssl.

LEDE changed px5g a few months ago with https://github.com/lede-project/source/commit/a7f6dc9f8b9602441a1b56bbe73d80711e2c640f to to have a new mbedtls library variant for px5g in addition to the old polarssl SSL variant. At that point the old px5g was renamed to px5g-polarssl and the other variant px5g-mbedtls. However, in LuCI the SSL dependency in both by luci-ssl and luci-ssl-openssl collections uses the short "px5g", which package does not exist any more in LEDE but does exist in Openwrt. Changing the dependency to px5g-polarssl would break things for Openwrt users.

As the build time "make defconfig" throws no complaints, I guess that as both px5g-polarssl and px5g-mbedtls do "provide" px5g, one of them gets selected at the build time config.

Manual installation of luci-ssl and luci-ssl-openssl may thus be broken in LEDE, at the moment. And that can't be fixed in LuCI without then breaking the install in Openwrt. The easiest solution might be to rename px5g-polarssl variant back to px5g in LEDE.

Ps. To my knowledge, the package uhttpd-mod-tls is not needed any more, as uhttpd is built with TLS support by default.

Pps. There is no need to have the @ targeting on the issue title. When the @ target is in the message body, the person gets the notification message about the issue and that is enough.

hnyman, 2016-10-04:
Well, it is a bit strange. I can't trigger your exact error in my build.  Buildbot has built luci-ssl-openssl also for your platform, so I think that there may something on your config or build system that triggers your exact problem. (Have you run "make defconfig" or "make menuconfig" after editing .config? If not, please do that before building.)

I noticed yesterday that when I installed luci-ssl-openssl via opkg to a live LEDE router, it installed px5g-mbedtls, which surprised me as it should have installed px5g-polarssl. The confusion comes from the reason that there is no default defined for the px5g variant. Likely there should be more clearly just a px5g package plus the mbedtls variant.

hnyman, 2016-10-04:
@jow- 
I am not sure if this should be fixed, but apparently the px5g-mbedtls creating commit https://github.com/lede-project/source/commit/a7f6dc9f8b9602441a1b56bbe73d80711e2c640f from cshore did not explicitly define one of the variants as default. Due to that the package dependency logic seems to pick the alphabetically first variant as the default, in this case mbedtls. I don't think that has been the intention, as I think that polarssl should still be the default.

From my tmp/.packagedeps:
```
package-$(CONFIG_PACKAGE_px5g-mbedtls) += utils/px5g
$(curdir)/utils/px5g/default-variant := mbedtls
$(curdir)/utils/px5g/variants += $(if $(CONFIG_PACKAGE_px5g-mbedtls),mbedtls)
package-$(CONFIG_PACKAGE_px5g-polarssl) += utils/px5g
$(curdir)/utils/px5g/variants += $(if $(CONFIG_PACKAGE_px5g-polarssl),polarssl)
```
For 99% of users that does not matter, as px5g is only selected by luci-ssl collections. There the px5g-polarssl variant seems to get selected first, likely due to libustream-polarssl being selected first. (but that px5g-polarssl selection seems to happen also for the luci-ssl-openssl collection, which I wonder)

px5g-mbedtls gets installed if you manually say just "opkg install px5g" in LEDE, where only px5g-polarssl and px5g-mbedtls packages actually exist. Then the default variant mbedtls gets installed, as the opkg logic seems to find it as the first of the two packages that "provide" px5g.

I am not sure if this should be fixed, and if the best solution would be
* a third "px5g" variant that would be default and would currently use polarssl
* px5g-polarssl being defined as the default variant
* px5g-polarssl being renamed to just px5g

Ps. Additionally I noticed a possible error in the ustream-ssl package. It has two default variants...
https://github.com/lede-project/source/blob/master/package/libs/ustream-ssl/Makefile#L54

Possibly the copy-paste used to create the mbedtls variant has included that DEFAULT_VARIANT line by accident and also the DEPENDS in mbedtls variant looks different than the other two optional variants: https://github.com/lede-project/source/commit/d9494cdf6dab6cf4c1b74f4031e9492d31f1c3dc

When I look at my tmp/.packagedeps, that does not seem to hurt much, as polarssl gets evaluated last and survives as the default variant.

This does not probably hurt users much, but might still be straightened up.

build000, 2016-10-04:
> (...)(Have you run "make defconfig" or "make menuconfig" after editing .config?(...)

No, I never editing .config and after run "make menuconfig" or/and "make defconfig" - only:
1.) git clone [LEDE]
2.) make package/symlinks V=s

hnyman, 2016-10-04:
I just tested cloning a new repo, updating & installing the feeds, and selecting luci-ssl-openssl via menuconfig. Works ok.
```
git clone https://git.lede-project.org/source.git
cd source/
./scripts/feeds update
./scripts/feeds install -a
make menuconfig
```
After selecting luci-ssl-openssl in menuconfig, px5g-polarssl was properly selected in .config.

The only way I have been able to reproduce your error is to manually edit .config to activate luci-ssl-openssl there and then start a build without first fixing the dependencies with make defconfig.

To me this still sounds like an user error, as the buildbots compile the package happily, too.

build000, 2016-10-04:
I downloaded the new source, installed feeds ... but ... how to use the old .config, you do not need to be set all over again ??? (For me apart from the sources and feeds there are symlinks to the local directory with packages and small changes in comgt, dnscrypt-proxy, additional scripts in the package/base-files/files/sbin, additional firmware for USB3, etc.).
So I used the "make defconfig" hoping that it will update my .config automatically - if not so ... how?

hnyman, 2016-10-04:
"make defconfig" basically just checks the dependencies starting from your current selections. the same is run when you use make menuconfig and save settings at the end.

> how to use the old .config

I use a .config recipe ".config.init" that contains only the selection that I have done. I copy the short recipe as the new .config and then I use "make defconfig" to expand it to the full config.  See https://forum.openwrt.org/viewtopic.php?pid=301814#p301814 for explanation

To create such a recipe, you should start with the output of ``` ./scripts/diffconfig.sh ```.  It shows the non-standard selections that you have currently made (or that have been enabled via dependencies due to your selections). That serves as the baseline and is much much more useful than the whole .config 

You should then prune from that those automatic dependencies that have been pulled in due to your actual selections. E.g. selecting luci selects uhttpd, luci-mod-admin-full etc. Keeping just the short recipe enables you to apply new config defaults & options to your old selections.

You need to rebuild .config rather regularly. Otherwise it starts accumulating unnecessary crust and changes in defaults go unnoticed, leading you into rather non-standard build.

build000, 2016-10-04:
@hnyman 
On the one hand, what you write is understandable and logical ... but highly impractical, confusing and cumbersome to perform "live" - perhaps easier I will torture ustwiając all over again - as far as possible, of course ... what I start slightly doubt ... (but clearly in need configurations is uhttpd-mod-tls-openssl - too much further configuration in the main system depends precisely on this choice).

hnyman, 2016-10-04:
It is actually rather simple, once you create the recipe. 

I am closing this issue, because it sounds very likely that this is not an actual error in the build system sources or logic. worksforme.

closed 2016-10-04 by hnyman