openwrt/luci #815: luci-app-commands error build/compile (i18n) - identical bug to luci-app-firewall and probably some other apps in LuCI

build000, 2016-10-04:
make -j8 BUILD_LOG=1

cat [sources]/logs/package/feeds/luci/luci-app-commands/compile.txt:
```
Packaged contents of /home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/luci-app-commands/ipkg-all/luci-app-commands into /home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/bin/packages/powerpc_464fp/luci/luci-app-commands_git-16.278.30451-e3287d8-1_all.ipk
bash: po2lmo: command not found
../../luci.mk:228: recipe for target '/home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/bin/packages/powerpc_464fp/luci/luci-i18n-commands-pl_git-16.278.30451-e3287d8-1_all.ipk' failed
make[3]: *** [/home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/bin/packages/powerpc_464fp/luci/luci-i18n-commands-pl_git-16.278.30451-e3287d8-1_all.ipk] Error 127
```

jow-, 2016-10-04:
@build000 - is your LuCI feed already including f7639bcd6a2caf7cc0f03e85744906c577eefeab ? In doubt please issue an `./scripts/feeds update luci` followed by `make package/luci-base/host/{clean,compile,install} V=s` - thne attmpt to build again-

build000, 2016-10-04:
yes - including - source sucked standard "git clone [source LEDE]" and installed feeds "make package / symlinks V = s" (~15-20 min. ago)

hnyman, 2016-10-04:
The git commit hash e3287d8 in the report looks to be up-to-date, as that is the merge commit of the fix in #813 by @NeoRaider 

i18n builds are failing also on buildbot due to missing po2lmo:
https://downloads.lede-project.org/snapshots/faillogs/mips_24kc/luci/

So apparently the fix in #813 is not yet good enough.

build000, 2016-10-04:
@jow- 
```
./scripts/feeds update luci
Updating feed 'luci' from 'https://git.lede-project.org/project/luci.git' ...
Already up-to-date.
Create index file './feeds/luci.index' 
#######################################
make package/luci-base/host/{clean,compile,install} V=s
make[1]: Entering directory '/home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE'
make[2]: Entering directory '/home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/feeds/luci/modules/luci-base'
rm -rf /home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/host/LuaSrcDiet-0.12.1 /home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/stamp/.luci-base_installed /home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/host/LuaSrcDiet-0.12.1/.built
make[2]: Leaving directory '/home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/feeds/luci/modules/luci-base'
make[1]: Leaving directory '/home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE'
make[1]: Entering directory '/home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE'
make[2]: Entering directory '/home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/feeds/luci/modules/luci-base'
. /home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/include/shell.sh; bzcat /home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/dl/LuaSrcDiet-0.12.1.tar.bz2 | tar -C /home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/host/LuaSrcDiet-0.12.1/.. -xf - 
touch /home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/host/LuaSrcDiet-0.12.1/.preparedd105eda54a64359bb1c5d0203cb353a3
touch /home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/host/LuaSrcDiet-0.12.1/.configured
make -C src/ clean po2lmo
make[3]: Entering directory '/home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/feeds/luci/modules/luci-base/src'
rm -f po2lmo parser.so version.lua *.o
cc    -c -o po2lmo.o po2lmo.c
cc    -c -o template_lmo.o template_lmo.c
cc  -o po2lmo po2lmo.o template_lmo.o
make[3]: Leaving directory '/home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/feeds/luci/modules/luci-base/src'
make -C /home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/host/LuaSrcDiet-0.12.1 bin/LuaSrcDiet.lua
make[3]: Entering directory '/home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/host/LuaSrcDiet-0.12.1'
make[3]: 'bin/LuaSrcDiet.lua' is up to date.
make[3]: Leaving directory '/home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/host/LuaSrcDiet-0.12.1'
touch /home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/host/LuaSrcDiet-0.12.1/.built
install -d -m0755 /home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/bin
install -m0755 src/po2lmo /home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/bin/po2lmo
install -m0755 /home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/host/LuaSrcDiet-0.12.1/bin/LuaSrcDiet.lua /home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/bin/LuaSrcDiet
mkdir -p /home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/stamp
touch /home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/host/LuaSrcDiet-0.12.1/.built
touch /home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/stamp/.luci-base_installed
make[2]: Leaving directory '/home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/feeds/luci/modules/luci-base'
make[1]: Leaving directory '/home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE'
make[1]: Entering directory '/home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE'
make[2]: Entering directory '/home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/feeds/luci/modules/luci-base'
. /home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/include/shell.sh; bzcat /home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/dl/LuaSrcDiet-0.12.1.tar.bz2 | tar -C /home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/host/LuaSrcDiet-0.12.1/.. -xf - 
touch /home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/host/LuaSrcDiet-0.12.1/.preparedd105eda54a64359bb1c5d0203cb353a3
touch /home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/host/LuaSrcDiet-0.12.1/.configured
make -C src/ clean po2lmo
make[3]: Entering directory '/home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/feeds/luci/modules/luci-base/src'
rm -f po2lmo parser.so version.lua *.o
cc    -c -o po2lmo.o po2lmo.c
cc    -c -o template_lmo.o template_lmo.c
cc  -o po2lmo po2lmo.o template_lmo.o
make[3]: Leaving directory '/home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/feeds/luci/modules/luci-base/src'
make -C /home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/host/LuaSrcDiet-0.12.1 bin/LuaSrcDiet.lua
make[3]: Entering directory '/home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/host/LuaSrcDiet-0.12.1'
make[3]: 'bin/LuaSrcDiet.lua' is up to date.
make[3]: Leaving directory '/home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/host/LuaSrcDiet-0.12.1'
touch /home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/host/LuaSrcDiet-0.12.1/.built
[... cut ...]
mkdir -p /home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/stamp
touch /home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/build_dir/target-powerpc_464fp_musl-1.1.15/host/LuaSrcDiet-0.12.1/.built
touch /home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/staging_dir/target-powerpc_464fp_musl-1.1.15/host/stamp/.luci-base_installed
make[2]: Leaving directory '/home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE/feeds/luci/modules/luci-base'
make[1]: Leaving directory '/home/sj/Pulpit/Projekty/WNDR4700/LEDE/LEDE'
```

hnyman, 2016-10-04:
I tested LEDE build with a luci-base Makefile that had both the new installation commands from f7639bcd6a2caf7cc0f03e85744906c577eefeab and also the original commands (before #812 and #813). There is a clear difference into where the po2lmo gets placed during the same build run.

NEW INSTALL COMMANDS:
```
	$(INSTALL_DIR) $(HOST_BUILD_PREFIX)/bin
	$(INSTALL_BIN) src/po2lmo $(HOST_BUILD_PREFIX)/bin/po2lmo
	$(INSTALL_BIN) $(HOST_BUILD_DIR)/bin/LuaSrcDiet.lua $(HOST_BUILD_PREFIX)/bin/LuaSrcDiet
---
install -d -m0755 /Openwrt/lede/staging_dir/target-mips_24kc_musl-1.1.15/host/bin
install -m0755 src/po2lmo /Openwrt/lede/staging_dir/target-mips_24kc_musl-1.1.15/host/bin/po2lmo
install -m0755 /Openwrt/lede/build_dir/target-mips_24kc_musl-1.1.15/host/LuaSrcDiet-0.12.1/bin/LuaSrcDiet.lua /Openwrt/lede/staging_dir/target-mips_24kc_musl-1.1.15/host/bin/LuaSrcDiet
```
OLD INSTALL COMMANDS:
````
	$(INSTALL_DIR) $(STAGING_DIR)/host/bin
	$(INSTALL_DIR) $(STAGING_DIR_HOST)/bin
	$(INSTALL_BIN) src/po2lmo $(STAGING_DIR_HOST)/bin/po2lmo
	$(INSTALL_BIN) $(HOST_BUILD_DIR)/bin/LuaSrcDiet.lua $(STAGING_DIR)/host/bin/LuaSrcDiet
---
install -d -m0755 /Openwrt/lede/staging_dir/target-mips_24kc_musl-1.1.15/host/bin
install -d -m0755 /Openwrt/lede/staging_dir/host/bin
install -m0755 src/po2lmo /Openwrt/lede/staging_dir/host/bin/po2lmo
install -m0755 /Openwrt/lede/build_dir/target-mips_24kc_musl-1.1.15/host/LuaSrcDiet-0.12.1/bin/LuaSrcDiet.lua /Openwrt/lede/staging_dir/target-mips_24kc_musl-1.1.15/host/bin/LuaSrcDiet
```
``` $(STAGING_DIR_HOST) ``` is not equal to ``` $(STAGING_DIR)/host ```. The changes in #812 and #813 assume that it has been the same.

@jow- @NeoRaider 
Not quite sure what is the correct fix. To revert back to the old code? Change the expected location of po2lmo where it is used for i18n packages? Or change po2lmo install commands?

NeoRaider, 2016-10-04:
Moving po2lmo was intentional, there should be a good reason to install anything to '$(STAGING_DIR_HOST)'.

Both `$(STAGING_DIR_HOST)` and `$(STAGING_DIR)/host` should be in PATH for package builds, I'm not sure what is going wrong here. Maybe it's a SDK-specific issue, I can't reproduce it using the normal build root. I'll investigate further...

hnyman, 2016-10-04:
I used normal LEDE buildroot (where I have done make dirclean earlier today) and got the error in a clean build after enabling a language for Luci. The test above was also done a in the same normal buildroot. So it is not an SDK issue, I think.

NeoRaider, 2016-10-04:
Ugh, noticed my mistake, I was testing against the wrong LuCI tree. I think I'll be able to proproduce it now...

NeoRaider, 2016-10-04:
Okay, the issue is that `Build/Exports` in `include/package.mk` is not applied to `Package/.../Install`, so `$(STAGING_DIR)/host` is indeed not availble during the install step.

I agree that it would be best to revert my two commits for now, until we have solved that.

build000, 2016-10-05:
fixed ... for now https://github.com/openwrt/luci/commit/e3287d84e107c4316fbf770a21b3088bc623beb5
issue to by closed

closed 2016-10-05 by build000