openwrt/luci #820: Luci-ssl-openssl

thagabe, 2016-10-07:
When this option is selected I get a compiler error since libustream-polarssl is selected even when I had selected luci-ssl-openssl. Maybe we should have luci-ssl-openssl automatically turn off libustream-polarssl.

hnyman, 2016-10-07:
Some other package then selects also libustream-polarssl. (Or it was already selected in your config.)
Make sure that if you have selected uhttpd-mod-tls, which is not actually needed, the openssl variant is used.

How did you select luci-ssl-openssl? Starting from an empty .config or modifying an existing one?

hnyman, 2016-10-07:
@jow- 
Do you see any actual need for uhttpd-mod-tls_* any more? uhttpd is currently compiled always with the TLS support active (since https://dev.openwrt.org/changeset/35295 in 2013) , so the only function for uhttpd-mod-tls_* is to select a libustream variant. But as we currently select a libustream variant directly in both luci-ssl and luci-ssl-openssl, there is is no actual need for uhttpd-mod-tls. It only leads into forcing users making the same encryption selection twice (if they select uhttpd-mod-tls in menuconfig).

uhttpd-mod-tls_* may help the user in the rare case where he selects uhttpd and wants TLS, but does not select Luci at all.

Or should the dependencies between luci-ssl* , uhttpd-mod-tls_* and libustream-* be reorganised somehow? Add a conditional dependency ``` @uhttpd:uhttpd-mod-tls_polarssl ``` to luci-ssl and ``` @uhttpd:uhttpd-mod-tls_openssl ``` luci-ssl-openssl ????  (Or even a direct dep ``` uhttpd-mod-tls_polarssl ``` or ``` uhttpd-mod-tls_openssl ```, as luci itself selects uhttpd in any case.) I think that it might decrease possibilities for conflicts.

(I haven't included uhttpd-mod-tls_* in my own build for ages.)

build000, 2016-10-07:
> Or should the dependencies between luci-ssl* , uhttpd-mod-tls_* and libustream-* be reorganised somehow? Add a conditional dependency @uhttpd:uhttpd-mod-tls_polarssl to luci-ssl and [...]

And this is probably the case glu - this change is necessary in the proposed form or another.
It is known that uhttpd-mod-tls with manual option is selected openssl/polarssl must be because as he rightly noticed are urzytkownicy (myself as I build on some devices), they do not use at all LuCI, while the need uhttpd + openssl exists independently of LuCI.

cshoredaniel, 2017-01-17:
polarssl removed from LEDE, probably such changes will occur in OpenWrt; in any event not much point to messing with this in LuCI.  Also mod-tls has been removed in at lease LEDE.

closed 2017-01-17 by cshoredaniel

psyborg55, 2017-01-24:
which one of these modules is responsible for Error 113 (net::ERR_SSL_VERSION_OR_CIPHER_MISMATCH): Unknown error. on older browsers?

hnyman, 2017-01-25:
> which one of these modules is responsible for Error 113 (net::ERR_SSL_VERSION_OR_CIPHER_MISMATCH): Unknown error. on older browsers?

Googling for that error reveals mostly stuff from 2011-2013:
https://www.google.fi/search?q=Error+113+(net%3A%3AERR_SSL_VERSION_OR_CIPHER_MISMATCH)%3A

Based on those, it is Chrome-specific. Might be old/stale/incompatible SSL certs in the browser cache, and when the SSL library has been updated, some of those old certs are not compatible any more. (Some of the older/weaker/vulnerable algorithms are disabled in the current SSL libs.)

psyborg55, 2017-01-25:
i had to enable uhttpd-mod-tls from Network -> Web Servers/Proxies menuconfig to make this work again