openwrt/luci #824: luci-app-commands public links wrong

muddyfeet, 2016-10-11:
Using the luci-app-command part of luci works well for the most part. However, there is an error in the links for public commands.

When hitting the “Link” button on the dashboard page, it returns a link in the form below:

“Access command with http://192.168.1.1/cgi-bin/luci/admin/system/commandscommand/cfg089944”

It should read:

“Access command with http://192.168.1.1/cgi-bin/luci/command/cfg089944”

The error appears to be in /usr/lib/lua/luci/view/commands.htm at line 113

`var link = location.protocol + '//' + location.hostname +
		           (location.port ? ':' + location.port : '') +
				   location.pathname.split(';')[0] + 'command/' +
				   id + (args ? '/' + args : '');
`
		
It perhaps should read (and please pardon my lack of programming ability...)

`var link = location.protocol + '//' + location.hostname +
		           (location.port ? ':' + location.port : '') +
				   location.pathname.split('luci/')[0] + 'luci/command/' +
				   id + (args ? '/' + args : '');`

cshoredaniel, 2017-01-14:
@muddyfeet Does this PR (https://github.com/openwrt/luci/pull/768) also resolve the issue (along with the main purpose of allowing text viewing instead of have to download the output).

muddyfeet, 2017-01-17:
Sorry for the delay responding - no, the code in PR (#768) doesn't fix the URI as you've already determined.  However, the changes proposed are a great improvement in other areas.

yousong, 2018-05-07:
Fixed by e4f77ac

closed 2018-05-07 by yousong