openwrt/luci #828: Dynamic tag broken

jp-bennett, 2016-10-16:
As reported in https://github.com/openwrt/packages/issues/3325 a package is showing errors in luci.  This package has worked perfectly in the past.  It sems that marking a section as Dynamic triggers the problem.  I wonder if @jow- has any thoughts on the issue.

cshoredaniel, 2016-12-09:
@jp-bennett @dkadioglu @jow- @hnyman  Would it be possible to rework the app along the lines of e.g. openvpn or create a table with options and type in fwknop and use a loop in the CBI model to create the fields?  If it's not working and hasn't been for some time, and there are so few uses that it hasn't come up before, I'm not sure spending time on making dynamic fields work again makes sense.

jp-bennett, 2016-12-10:
@cshoredaniel It works without the dynamic fields, but is a bit clunky with a whole bunch of options.  There are about 5 commonly used options, but a whole bunch of other options that a user *might* want to use, hence supporting the Dynamic fields.  I'll look into other the other options you mentioned.

If dynamic fields aren't going to be fixed, then they should be removed altogether, rather than left in a broken state.

cshoredaniel, 2016-12-10:
@jp-bennett Someone may work on them, if there is sufficient need/demand + time, but if, as it appears, the functionality is little used I think @jow- and @hnyman probably would rather reduce maintenance burden for themselves.  But they will speak for themselves I'm sure.

jow-, 2016-12-10:
I'd vote for getting rid of dynamic options as well. In their current form they're a usability nightmare.
I usually replaced dynamic options in other tabs by dividing the form into a common or general tab and and advanced tab holding all the exotic options.

jp-bennett, 2016-12-10:
@jow- Fine with me, I can do without the option with the general/advanced approach.  I say pull the plug, rather than leave them broken.

cshoredaniel, 2016-12-10:
@jp-bennett Do you need help with that?  I've got a few other things on my plate, but I can see about assisting with the transition for fwknop.

cshoredaniel, 2017-01-14:
@hnyman @jow- @chris5560 Since dynamic tags are currently broken anyway and the only users left are asterisk and mod-mini, how do you fee about my at least marking those as broken for the upcoming release and removing the dynamic tags from cbi.lua?

cshoredaniel, 2017-01-17:
@hnyman @jow- Do we need to fix dynamic tags for 17.01 (for -mini and -asterisk) or can we mark those as broken (those are currently the only users not marked broken; I think pbx also used them).  In any event mini and asterisk are not currently functional.

hnyman, 2017-01-22:
> Do we need to fix dynamic tags for 17.01 (for -mini and -asterisk) or can we mark those as broken (those are currently the only users not marked broken; I think pbx also used them). In any event mini and asterisk are not currently functional.

Marking -mini and -asterisk broken in 17.01 is ok for me. If they do not work, it makes no sense to build them.

@jow-   ?