openwrt/luci #829: OpenVPN Single Config File Support

gh0stl4b, 2016-10-20:
Please add support for just a single file configuration, the ones that end in .ovpn, I think many people will find this feature very good to have. Especially if they like me use VPNGate alot and have a lot of devices on the network.

Thank you in advance.

jow-, 2016-10-20:
Just rename it to *.conf and place it as n /etc/openvpn/ - no further configuration needed.

gh0stl4b, 2016-10-20:
Yes, but then it does not show up in the web interface... And after placing the config file in that directory absolutely nothing happened, I'm still on my providers line. (And yes I know for a fact that the host is up and running)

jow-, 2016-10-20:
It needs to end in .conf and you need to run /etc/init.d/openvpn restart to make openvpn pick it up. Also ensure that it is in unix line ending format.

howels, 2016-11-07:
I can confirm, the VPN process is started but the luci GUI does not show a new VPN connection, which is confusing and means that SSH is needed to start/stop the connection.

gh0stl4b, 2016-11-12:
I've done everything exactly as I've been told here, after testing my IP address several times I have not noticed any change, it shows me my provider, but when I use the same config file for Viscosity it connects instantly and shows me the IP of the server. So IDK what to do, maybe it wold be better if the devs could add a specific field in which you can upload a .ovpn file and the code will automatically do the rest for you, as well as add the connection to the list.

rgb03, 2016-12-18:
For me it worked after referencing the file in /etc/config/openvpn.  I didn't even need to change the .ovpn extension. Suppose your single file is client1.ovpn:
1. Place the file as /etc/openvpn/client1.ovpn
2. edit /etc/config/openvpn and add the following:
```
config openvpn 'client1'
    option config '/etc/openvpn/client1.ovpn'
    option enabled '1'
```

cshoredaniel, 2017-01-17:
@gh0stl4b @howels @rgb03 Can you confirm whether

1) rgb03's config shows up in the advance config screen (i.e. can be created/edited by LuCI (since this is the LuCI issues page))
2) with that config status is correctly shown.

Wedmer, 2017-08-29:
I can add uploading custom config but I need feedback on how other options should be used.

ktaragorn, 2018-07-24:
This is so wierd.. on top of adding the custom block to the file, I am trying to rename one of the samples. If i rename to the same as my custom block, it dissapears. It shows up otherwise..

ktaragorn, 2018-07-24:
Hmmm.. is the word ZenVPN blacklisted here?? If i rename it to anything else it appears to show up.

cshoredaniel, 2018-07-29:
You can't have two sections with the same name.  If you do the latest one in the config file wins.

dibdot, 2018-10-23:
now in trunk with 0b04912

closed 2018-10-23 by dibdot