openwrt/luci #835: Luci-app-statistics rrd graph titles omit interface name

gfokkema, 2016-10-26:
Was trying to post this on the forums, since this might be more of a device specific thing.
However, since links are not allowed there, this will be much easier (and it might not be device specific after all).

I'm configuring rrdtool / graphs on a new openwrt router.
After having created a working config to get rrdtool graphs to display, all graphs from the 'interfaces' plugin accessible at `/cgi-bin/luci/admin/statistics/graph/interface` don't show an interface name.
Instead their title is displayed as:
> \<hostname\>: Transfer on

instead of
> \<hostname\>: Transfer on eth0

So i've traced down the issue and noticed that the title is passed to rrdtool incorrectly in [rrdtool.lua](https://github.com/openwrt/luci/blob/master/applications/luci-app-statistics/luasrc/statistics/rrdtool.lua#L562) since the variable `plugin_instance` seems to be empty.

I've traced that down to the variable `inst` being empty in [luci_statistics.lua](https://github.com/openwrt/luci/blob/master/applications/luci-app-statistics/luasrc/controller/luci_statistics/luci_statistics.lua#L186), but that's as far as I've gotten.

Can someone point me in the right direction as to why this variable is not filled with the interface name?
My config:
[luci_statistics.txt](https://github.com/openwrt/luci/files/552050/luci_statistics.txt)

Thanks in advance.

hnyman, 2016-10-26:
Which Openwrt version (and Luci version) you are using?

Interface graphs' titles work ok for me with the build from up-to-date master. (I adjusted the config via Luci to select multiple interfaces to mimic your setup.)

JamesT42, 2016-11-03:
@gfokkema Are you also using a Turris Omnia? Because I have the same problem on that device. Unfortunately they have got their own fork of luci going, which doesn't exactly help with bug tracking...

cshoredaniel, 2016-12-09:
@gfokkema up to date master works for me as well; can you confirm the version and of LuCI and/or openwrt?

cshoredaniel, 2017-01-14:
No response for a month; appears to be a third party modification; closing

closed 2017-01-14 by cshoredaniel