openwrt/luci #837: luci.model.network not documented

karlp, 2016-10-27:
The autogenerated API is missing all evidence of the luci.model.network package.

cshoredaniel, 2016-12-09:
There are no luadoc comments in https://github.com/openwrt/luci/blob/master/modules/luci-base/luasrc/model/network.lua.  As the plan is to move to luci-ng, old style luci documentation is unlikely to get a lot of attention unless a community member submits a PR.

jow-, 2021-12-03:
The Lua class is more or less deprecated and has no active users anymore afair. The Javascript counterpart is documented at http://openwrt.github.io/luci/jsapi/LuCI.network.html

closed 2021-12-03 by jow-

karlp, 2021-12-06:
well, grep for luci.model.network still shows plenty of references, even in master, let alone 21.02, but yeah, I know it's on it's way out.  :)