openwrt/luci #838: Table breaks out of layout (cbi/tblsection)

danrl, 2016-10-30:
Working on luci-proto-wireguard. Unfortunately, the peers require quite some options. Too many for the table, so the table breaks out of the layout. Am I doing it wrong or is this an issue that needs to be adressed by e.g. breaking the rows somehow? I don't think there is an easy fix for that. Browser don't support automatic wrapping/breaking of TRs and teaching LUA how and when to split long TRs into multiple TRs seems tedious. Also, how to split the TH row so that the columns still match the corresponding TDs in TBODY? Ideas?

See the ugliness here :)

[image]

cshoredaniel, 2016-12-09:
@danrl Likely some of the fields in the table have fixed sizes instead of based on the size of the browser window (e.g. public key, my address, etc entry fields) which cause the table to be unable to resize to the browser window.  Might be theme-specific issue too.

danrl, 2016-12-09:
This has been resolved (for me). I chose a more relaxed design approach in `luci-proto-wireguard` eventually.

closed 2016-12-09 by danrl