openwrt/luci #839: wifi scan function broken

psyborg55, 2016-11-03:
running R49946 while client device is connected scanning for other networks works as expected, if the connection drops because of for example remote device reboot scanning does not work but displays empty page.

jow-, 2016-11-03:
I do not understand the report. Isn't it normal that the pag does not load if the connection drops?

psyborg55, 2016-11-03:
sorry, should've make it more clear, client device on openwrt that connects to upstream network not client device that connects to openwrt.

this has always worked with previous revisions. screenshot added:

[image]<br /><a target='_blank' href='https://postimage.org/'>host image</a><br /><br />

cshoredaniel, 2016-12-09:
@psyborg55 Are you saying that the issue is that if the upstream device reboots while doing a wireless scan on the device running openwrt that you get an empty wireless scan page?  If so, this is most likely an issue with the underlying wireless driver in which the disconnection resets the associated interface.

psyborg55, 2016-12-10:
as soon as client connection to the upstream AP is lost the scan function returns this empty page. while connected it's listing networks as usual.

cshoredaniel, 2016-12-21:
@jow- @hnyman  Any way of confirming if this is a driver issue?  It sounds to me like when connection dies the driver resets and therefore scan fails.

hnyman, 2016-12-22:
Well, to my knowledge LuCI contains no wifi scanning logic by itself. All scanning is done by the wifi driver and LuCI just parses the output of the driver scanning command, which I think is "iw wlan0 scan" (or wlan1) and displays that output. 

The scanning works also from console. If iw fails also there, there is no way LuCI could fix that.
```
root@lede:~# iw wlan0 scan
BSS 10:bf:48:33:22:39(on wlan0)
        TSF: 711625688938 usec (8d, 05:40:25)
        freq: 5220
        beacon interval: 100 TUs
        capability: ESS (0x0001)
        signal: -79.00 dBm
        last seen: 3840 ms ago
        SSID: ASUS_5G
...
```

In general it sounds strange that scanning would be dependent on existing connections, as the purpose of scanning is to find new available networks.

Which router and wifi driver we are talking about?

psyborg55, 2016-12-22:
xiaomi mini, happens with both ar9271 and rt3070 usb devices. after the connection is lost (remote AP restart) scan doesn't work and as soon as it's re-established (remote AP up) scan works again.

cshoredaniel, 2017-01-14:
@psyborg55 can you test iw and iwinfo scans and see it happens from the command line too?  It doesn't seem like a LuCI issue but core.

psyborg55, 2017-01-15:
got this:

```
root@OpenWrt:~# iwinfo wlan2 scan
No scan results

```

when it works, the iwinfo scan takes 15 seconds and luci scan 45 seconds - there's for sure something broken

cshoredaniel, 2017-01-16:
luci scan time is just UI load time I think (yeah LuCI has performance issues; hence the desire for LuCI-ng); if iwinfo is not working then it's a driver issue and not a LuCI issue.

closed 2017-01-16 by cshoredaniel

dji-transpire, 2018-04-14:
This is indeed an issue, Luci scans, or tries to do so, and hangs. on the command line, it works.

iw wlan0 scan is fine, the GUI command hangs and does not produce results

chaos calmer 15.05 on a HooToo TM06 (with GL-TM300N firmware, since it uses the same ramips chipset)