openwrt/luci #842: Setup luci-app-hd-idle for multiple hdd?

schovi, 2016-11-05:
Is there any way, how to configure `luci-app-hd-idle` for more than one hard disk? Web UI definetely does not allow it and I cant find out, how to write it directly into config to make it work.

chris5560, 2016-12-01:
@schovi
It's currently not possible to add a new config section for more disks inside luci-app-hd-idle.
Inside /etc/config/hd-idle file copy the whole config section for each disk you need.
Go back to Luci and there are shown all sections and can be changed to your needs.

@gwlim
From my point of view luci-app-hd-idle and hd-idle package needs to be reworked because UCI normally not accept "-" dashes in config/option/list names. This might be the reason why you cannot add more entries inside luci-app-hd-idle.

schovi, 2016-12-02:
@chris5560 I tried something like that, but result was "unstable". I will give a change to it one more time.

schovi, 2016-12-03:
@chris5560 
/etc/config/hd-idle:
```
config hd-idle 'sda'
	option idle_time_unit 'minutes'
	option idle_time_interval '10'
	option disk 'sda'
	option enabled '1'

config hd-idle 'sdb'
	option idle_time_unit 'minutes'
	option idle_time_interval '10'
	option disk 'sdb'
	option enabled '1'
```

But even when I try everything (start, stop restart) only first one is applied and 

root@xxxxxxxxx:~# ps  | grep hd-idle
26474 root       692 S    /usr/bin/hd-idle -a sda -i 600
26561 root      1076 R    grep hd-idle

schovi, 2016-12-04:
@chris5560 Maybe problem is in `/etc/init.d/hd-idle` with 

```
start_service() {
	local section="$1"
	args=""
	config_get "interval" "$section" "idle_time_interval"
	config_get "unit" "$section" "idle_time_unit"
	append_string "$section" "disk" "-a"
	config_get_bool "enabled" "$section" "enabled" '1'
	[ "$enabled" -gt 0 ] || return 1
	service_start /usr/bin/hd-idle $args -i "$(compute_seconds $interval $unit)"
}
```

I can't find what exactly is `service_start` and how it behave. But problem seems to be there. It starts one hd-idle process and second is ignored. I put `echo` before that line and it get there with correct arguments, but just wont start. maybe service_start should have some kind of "name" to be correctly identified?

chris5560, 2016-12-04:
@schovi

hd-idle is working with unnamed sections, so please try to remove 'sda' / 'sdb'
from the line config hd-idle.
Hope this helps.

Otherwise you need to open an issue at openwrt/packages
Last known maintainer is @gwlim

bobafetthotmail, 2016-12-04:
the newer hd-idle package in the feeds supports multiple disks and schovi seems to have fixed the issue from CLI by copy-pasting the newer initscript over the older one.

schovi, 2016-12-05:
Thanks guys. I know little more, where to find sources and solutions now.

closed 2016-12-05 by schovi