openwrt/luci #844: Using 802.11r Fast Transition and LUCI causes the removal of /etc/config/wireless nasid keyword

muddyfeet, 2016-11-08:
Using 802.11r Fast Transition config - this requires hand editing /etc/config/wireless

Subsequently, making any change to the wireless config in LUCI (eg. channel) causes the nasid line to be deleted and the radio will not restart.

Problem observed with TP-Link Archer C7 v2, LEDE Reboot r2094 / LuCI Master (git-16.306.57050-d7d9772)

This issue seems to have been around for a while - see https://dev.openwrt.org/ticket/20764

I agree with the poster on on the openwrt ticket above that the issue seems to be a confusion between the RADIUS nasid and the 802.11r nasid. Maybe a solution is to change the 802.11r nasid keyword to something else?

/etc/config/wireless listed below:

config wifi-device 'radio0'
	option type 'mac80211'
	option hwmode '11a'
	option path 'pci0000:01/0000:01:00.0'
	option country 'GB'
	option txpower '20'
	option channel '108'
	option htmode 'VHT80'
	option log_level '0'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option encryption 'psk2+ccmp'
	option key 'probablynotthekey'
	option mode 'ap'
	option ssid 'notarealssid'
	option ieee80211r '1'
	**_option nasid '14DD20BE88C8'_**
	option mobility_domain '4f58'
	option r0_key_lifetime '10000'
	option r1_key_holder '14DD20BE88C8'
	option reassociation_deadline '1000'
	option pmk_r1_push '1'
	list r0kh '14:DD:20:47:14:E4,14DD204714E4,00112233445566778899aabbccddeeff'
	list r0kh '14:DD:20:47:14:E5,14DD204714E5,00112233445566778899aabbccddeeff'
	list r1kh '14:DD:20:47:14:E4,14:DD:20:47:14:E4,00112233445566778899aabbccddeeff'
	list r1kh '14:DD:20:47:14:E5,14:DD:20:47:14:E5,00112233445566778899aabbccddeeff'

cshoredaniel, 2016-12-09:
@muddyfeet This may involve ensuring changes for nasid for RADIUS doesn't conflict with Broadcom(?) wifi.  If netifd handles the manual uci config correct then it is in fact the LuCI model that is wrong, but requires building and testing on applicable hardware.
If you're willing and able to build yourself I can create a branch for you to test.

muddyfeet, 2016-12-12:
I've already done a bit of work on this on my local copy of wifi.lua.  Yes, the LuCI model is wrong rather than the underlying UCI model.  

There are two options that I can see, not deleting the nasid line if ieee80211r = 1 && (encryption = psk || encryption = psk2) or exposing the whole fast transition feature through LuCI.  I've opted for the later as it suits my need.

My feeling is that it probably is time to start exposing the fast transition features in LuCI.  I'm happy to have a go at it, although my programming ability is somewhat rusty - I learnt Fortran 77 and Pascal :)

cshoredaniel, 2016-12-13:
@muddyfeet Hey thanks for looking into this!  If you need help feel free to ask questions, and if I can I will help.  I like the exposing the fast transition mechanism from UCI option, if you're willing to give it a go.  I'm not quite as old a hand (later version of fortran, and started with C64 BASIC and assembler before high school, then Pascal, and C.  But have been programming since then, more time that and concentration (and other issues) that I have to contend with, that have been impacted by medical concerns (e.g. side effects of medications).

hnyman, 2017-02-08:
Just documenting that this is a real bug. I just tested with my new 802.11r setup. The nasid value gets deleted if I edit any option, as I have WPA2-PSK encryption (instead of WPA2-EAP).

In addition to https://dev.openwrt.org/ticket/20764 it has reported also as https://bugs.lede-project.org/index.php?do=details&task_id=274

Like said, the root reason is the usage of nasid uci option name for two separate purposes. And in LuCI the nasid keyword is currently dependent on encryption being WPA-EAP or WPA2-EAP instead of the personal usage WPA-PSK and WPS2-PSK
https://github.com/openwrt/luci/blob/master/modules/luci-mod-admin-full/luasrc/model/cbi/admin_network/wifi.lua#L903

The easy quick solution solution would be to
* remove encryption dependency from the option in LuCI, so that values would not be deleted just due to the encryption. That would be on line with the Openwrt commit from 2015, where nasid's connection to WPA-EAP was cut: https://github.com/openwrt/openwrt/commit/5ebd9314c0a31bd5896d1b7b9a0a2f48a7f9e9d6
* add a help note to the value mentioning that it is used for two different purposes.

@jow- 
Does that sound reasonable?  Any objection for removal of the encryption dependency?

@muddyfeet 
Do you have any 802.11r code ready for a PR?

hnyman, 2017-02-09:
I pushed a fix to LuCI master with https://github.com/openwrt/luci/commit/f02794f28c343070487bbc794b425f91b675fdb9

LuCI should be properly taught about 802.11r, but this should serve as the first aid.

muddyfeet, 2017-02-09:
I've been slowly working away learning lua and the luci code.

I've also just created a pull request with support for 802.11r fast transition.  It's not perfect as the R0KH & R1KH values are just a comma separated list as per the UCI file.

https://github.com/openwrt/luci/pull/994

If anyone wants to improve on it feel free - I think I've reached the limits of my capability.

closed 2017-02-09 by hnyman

cshoredaniel, 2017-02-09:
@muddyfeet thank you!  It looks like @hnyman has taken this up