openwrt/luci #845: OpenVPN interfaces doesn't show ip addresses

etomm, 2016-11-10:
In the Interfaces page all the interfaces tied to tun/tap devices doesn't show any ip address. This was not the case with CC 15.05.

[image]
[image]

As you can see in the first image there is nothing, instead in the second there are ip (the second is CC 15.05).

My /etc/config/network:
```
config interface 'loopback'
	option ifname 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'XXXX:XXXX:XXXX::/48'

config interface 'lan'
	option force_link '1'
	option type 'bridge'
	option ifname 'eth0.1'
	option proto 'static'
	option netmask '255.255.255.0'
	option ip6assign '64'
	option igmp_snooping '0'
	option metric '0'
	option stp '1'
	option ipaddr '192.168.0.1'

config device 'lan_dev'
	option name 'eth0.1'
	option macaddr 'XX:XX:XX:XX:XX:XX'

config interface 'wan'
	option ifname 'eth0.2'
	option _orig_ifname 'eth0.2'
	option _orig_bridge 'false'
	option proto 'pppoe'
	option username 'XXX'
	option password 'XXX'
	option ipv6 'auto'
	option metric '1'
	option peerdns '0'
	option dns '127.0.0.1'
	option keepalive '20 4'

config device 'wan_dev'
	option name 'eth0.2'
	option macaddr 'XX:XX:XX:XX:XX:XX'

config interface 'wan6'
	option ifname 'eth0.2'
	option proto 'dhcpv6'

config switch
	option name 'switch0'
	option reset '1'
	option enable_vlan '1'

config switch_vlan
	option device 'switch0'
	option vlan '1'
	option ports '1 2 3 4 0t'

config switch_vlan
	option device 'switch0'
	option vlan '2'
	option ports '5 0t'

config interface 'bridgetcp'
	option proto 'none'
	option auto '1'
	option ifname 'tun1'
	option igmp_snooping '0'

config interface 'vpn'
	option _orig_ifname 'tun2'
	option _orig_bridge 'false'
	option ifname 'tun2'
	option proto 'none'
	option auto '1'
	option delegate '0'
	option defaultroute '1'
	option metric '100'

config interface 'P2P'
	option proto 'none'
	option ifname 'tun3'
	option auto '1'
	option delegate '0'
	option defaultroute '1'
	option metric '90'

config interface 'bridgeudp'
	option ifname 'tun-br-vpnudp'
	option proto 'none'
	option auto '1'
	option igmp_snooping '0'

config interface 'setup'
	option proto 'static'
	option delegate '0'
	option ipaddr '192.168.254.1'
	option netmask '255.255.255.0'
```

bol-van, 2016-11-19:
Guys, the thing that is very wrong with openvpn is that it does not intergrate with netifd. I created my own netifd helpers so that its possible to ifup/ifdown openvpn interface. I find this approach much more compliant and comfortable.

cshoredaniel, 2016-12-09:
@bol-van Have you got them published (and what license?) anywhere?  It would be good to get openvpn into the modern age from the point of view of openwrt.

bol-van, 2016-12-10:
I can share my work but I dont pretend it to be the best quality and decisions made was perfect. I done it for myself.

jscinoz, 2017-04-03:
@bol-van Are you still willing to share your netifd helpers for openvpn? The lack of netifd integration is causing additional issues (openwrt/openwrt#220, openwrt/packages#3486).

bol-van, 2017-04-03:
No problem. I'll share my work. But I will not take part in making it clean and compliant, i've finished with programming and do it only for myself and dont share my code because it can be dirty and does not meet clean standards. Take it as is and do whatever you want.
https://www.sendspace.com/file/9vmqh4

My code relies on standard uci openvpn config in /etc/config/openvpn  but requires to make section "enabled 0" to stop it from launch by /etc/init.d/openvpn.
This approach leaves luci-app-openvpn working
Add proto section to /etc/config/network as in example

dibdot, 2018-11-25:
not a LuCI issue at all.

closed 2018-11-25 by dibdot

brianjmurrell, 2020-01-26:
> https://www.sendspace.com/file/9vmqh4

Is 404 unfortunately.