openwrt/luci #846: download files from luci

mirjafarabedi, 2016-11-12:
I've uploaded some files in my OpenWrt root , /etc , /tmp directories.
I want user to be able to download uploaded files from from the above directories .
How can i **give a download URL to user** , to download that files ?

hnyman, 2016-11-12:
You should not give them access to /etc and root directories, as then they can also see your settings and other private stuff.

By default LuCI allows read access to stuff under /www directory
You should place the public stuff there.

Note that you can utilise symlinks, as explained here:
https://forum.openwrt.org/viewtopic.php?pid=298615#p298615

> Just create a symlink to the directory you want to share under /www like this:
> 
> `# ln -s /path/to/shared/folder /www/sharename`
> 
> And you can access it like this: http://192.168.1.1/sharename

mirjafarabedi, 2016-11-12:
I know this security risk (download from /etc and ...) . But in my app i monitor some sensor information and save them in /tmp folder.(in memory). and then i want access user download this exported info after authenticating with his username and password.Is this not possible ? (Thanks in advance for your response)

mirjafarabedi, 2016-11-12:
Unfortunately your answer doesn't work for me

hnyman, 2016-11-12:
Works for me, but naturally only from LAN. If you want it to work from wan, you need to adjust firewall etc.

Below is example of outputting tail of kernel log to file /www/test/test.txt and then how it is visible in the browser:

```
root@lede:~# cd /www
root@lede:/www# mkdir test
root@lede:/www# cd test/
root@lede:/www# dmesg | tail > test.txt
root@lede:/www/test# cat /www/test/test.txt
[67087.447386] [<c021f9bc>] (handle_IPI) from [<c0209474>] (gic_handle_irq+0x90/0xbc)
[67087.454677] [<c0209474>] (gic_handle_irq) from [<c020a594>] (__irq_svc+0x54/0x70)
[67087.462130] Exception stack(0xdd471f80 to 0xdd471fc8)
[67087.469712] 1f80: 00000001 00000000 00000000 c020b460 dd470000 c07c0480 c06d0818 00000000
[67087.474754] 1fa0: c07ba224 dd471fd8 00000000 c07bc100 1d25e000 dd471fd0 c021a0f0 c021a0f4
[67087.482885] 1fc0: 60000013 ffffffff
[67087.491043] [<c020a594>] (__irq_svc) from [<c021a0f4>] (arch_cpu_idle+0x38/0x5c)
[67087.494356] [<c021a0f4>] (arch_cpu_idle) from [<c0260458>] (cpu_startup_entry+0x150/0x21c)
[67087.501989] [<c0260458>] (cpu_startup_entry) from [<4220952c>] (0x4220952c)
[67087.510132] ---[ end trace 829b74d04ace807a ]---
```
[image]
[image]

mirjafarabedi, 2016-11-12:
Thanks to answer.
But i have an other problem.
In LuCI web app in the system-backup/flash firmware mene user can download a backup of /etc folder . As below : 
[image]

How can i implement a functionality like this to allow user download files from /etc folder ?

hnyman, 2016-11-12:
> In LuCI web app in the system-backup/flash firmware mene user can download a backup of /etc folder

LuCI doesn't actual make a copy of /etc. It only launches a sysupgrade script that collects files from /etc (also also from other places if necessary) and creates an archive of those to /tmp (I think) and then sends that to the browser. Source code for the LuCI side functionality is here, in case you want to write something similar for your needs: https://github.com/openwrt/luci/blob/master/modules/luci-mod-admin-full/luasrc/controller/admin/system.lua#L302

The sysupgrade script is coming from base-files: https://github.com/openwrt/openwrt/blob/master/package/base-files/files/sbin/sysupgrade 

The easy approach for you is to use the /www like I showed, and just make sure that you have script that copies the files there as needed. If you want more complex access to other parts of the file system, feel free to start coding along the lines I showed here.

mirjafarabedi, 2016-11-12:
Thank you for your good answers . I will test your suggested solutions and I will announce the results.
Thanks again :))

mirjafarabedi, 2016-11-14:
I created a directory in my /etc folder . and then i linked it to /www/test (ln -s /etc/test1 /www/test)
and it works for me . 
Thank you very much :)

closed 2016-11-14 by hnyman