openwrt/luci #853: Keep values of dependent fields

maximeguerreiro, 2016-11-18:
Hello,
I use a `Flag` that allows me to enable/disable a whitelist, and this whitelist is stored using a `DynamicList`.
Thing is, when I disable the flag the list is lost. I've tried using `.rmempty = false` on my list, without success.

Sample code : 
```
use_whitelist = s:option(Flag, "use_whitelist","Use whitelist")
use_whitelist.enabled = "on"
use_whitelist.disabled = "off"
use_whitelist.default = use_whitelist.enabled
use_whitelist.rmempty = false

whitelist = s:option(DynamicList, "whitelist", "The list")
whitelist.optional = false
whitelist.datatype = string
-- whitelist.rmempty = false
whitelist.placeholder = "me"
whitelist:depends("use_whitelist", "on")
```

Is this the expected behavior, and is there a way to bypass it? For now I have commented out the `depends`.
Thanks! :smile:

cshoredaniel, 2016-12-09:
I seem to recall that using tracking_missing ( https://github.com/openwrt/luci/blob/6a11f71/modules/luci-base/luasrc/cbi.lua#L1301 ) in an option leaves the setting in the UCI config, but I could be remembering that wrong.

cshoredaniel, 2017-01-20:
@mguerreiro Have you the opportunity test 'tracking_missing'?

mguerreiro, 2017-01-23:
Hello. Sorry for the delay and thanks for pinging me.

Using `tracking_missing`, this behavior still occurs: the list is hidden from the web interface and not sent in the POST /config, so it gets removed server-side

cshoredaniel, 2017-01-23:
@mguerreiro thanks, will keep this open then and hopefully it can be looked into (I don't have time myself at the moment).

jow-, 2018-07-03:
This is by design. The best solution is to use both `.rmempty = false` and a custom `.write()` implementation that refuses to write empty data.

closed 2018-07-03 by jow-