openwrt/luci #854: luci-proto-wireguard: missing input fields

thinrope, 2016-11-18:
I just saw pull request #852 by @danrl and tested it out.
Unfortunately couldn't make it work :-|

At least one reason being that it lacks input fields to specify interface (local) IP address.
After configuration, one needs to give local IP address to the interface (and bring it up).
`ip addr add 192.168.xx.yyy/24 dev wg0`

See https://www.wireguard.io/quickstart/

(I am also new to wireguard, apologize, if there is some magic that is supposed to work) 

I have successfully configured a new interface from the shell with the following, where this(=OpenWRT box) has 192.168.6.113/24 address and the other (=Gentoo box) has 192.168.6.12/24 address:
````bash 
modprobe wireguard
ip link add dev wg0 type wireguard
wg genkey |wg set wg0 listen-port 2345 private-key /proc/self/fd/0 peer QfGI8PaVcUphJv4ntjIZHAZt4NXgA0bU3LHQvk0x/QI= endpoint 192.168.6.12:2345 allowed-ips 192.168.13.0/24
ip addr add 192.168.13.113/24 dev wg0
ip link set dev wg0 up
````
then configure the other host (use `wg show wg0 public-key` to get the public key needed to setup the other host).

So the minimum config for a wireguard interface should include local IP address and (peer OR mask), IMHO.
I had a look at the Lua, but am not sure how it works, sorry :-(

danrl, 2016-11-18:
Hi Kalin,

have you created a static addressing on top of the Wireguard interface?

Please see this article: https://www.danrl.com/2016/11/16/openwrt-luci-proto-wireguard.html

Let me know if there are any remaining issues.

Dan

thinrope, 2016-11-18:
Hello Dan,

Thank you for the quick reply! 
Nope, I hadn't seen your blog, had never occurred to me to try to define a new interface in luci ...
I read it and managed to configure it properly, apologies for the noise and not RTFM enough!

I guess I was misunderstanding the naming of "Interface" in luci then.

Is there any place your blog or relevant documentation can be found inside luci?
Kalin.

zx2c4, 2016-11-18:
Documentation: https://wiki.lede-project.org/docs/user-guide/tunneling_interface_protocols

thinrope, 2016-11-18:
Yep, that is even better!
I just never managed to compile LEDE (other issues) and am coming form OpenWRT land (and couldn't find docs there).
May be include a link to that UG page from the luci interface itself? Only few bytes shouldn't matter.

zx2c4, 2016-11-18:
Reading this blog post:

> I wanted to allow a wide range of setups and enable everyone to do even the weirdest things with their routers. So I removed the direct addressing feature that I was implemented in an earlier version. Luckily, you can create a static configuration on top of foo by creating a new device and selecting Static address as protocol.

I'm sort of curious what exactly you have in mind. How do you envision WireGuard working when it doesn't have a statically assigned IP address?

danrl, 2016-11-18:
> I guess I was misunderstanding the naming of "Interface" in luci then.

No worries, common mistake. Not a fan of "interface" either in this context.

danrl, 2016-11-18:
> I'm sort of curious what exactly you have in mind.

Have not tried it yet, but unicast DHCPv6 may work. Not yet thought through and also off-topic here I guess.

danrl, 2016-11-18:
@thinrope Not your fault, the documentation is only two days old :)

I am happy to see the code already being used.

zx2c4, 2016-11-18:
> Have not tried it yet, but unicast DHCPv6 may work.

How do you imagine this will work with AllowedIPs? The DHCP server dynamically fiddles with peers' entries? That'd be interesting.

I mention this here, because maybe there's some reason for adding back the static addressing to the script itself. But I can also understand why you prefer the abstraction.

danrl, 2016-11-18:
The static address support was rudimentary and did not reflect all of OpenWrts capabilities. Also, it would mean re-implementing what is already in OpenWrt. However, convince me it is worth the effort and I will dig into it.

cshoredaniel, 2016-12-09:
@zx2c4 @thinrope  Please reopen as a feature request (with convincing requested by @danrl ) if more support is needed.

closed 2016-12-09 by cshoredaniel