openwrt/luci #855: luci-proto-wireguard: wrong dependency

zorun, 2016-11-19:
@danrl luci-proto-wireguard should depend on `kmod-wireguard` and `wireguard-tools` instead of `wireguard`.  The latter is a virtual package.

hnyman, 2016-11-19:
But using the virtual package keeps dependencies in LuCI simpler, right? It does not hurt, as the dependency chain gets evaluated.

zorun, 2016-11-19:
The fact is, I'm not even sure it works right now :)

I don't see any reference to a `wireguard` package anywhere on the buildbots: it does not exist as a package and is not present in Packages.gz.

zorun, 2016-11-19:
See https://bugs.lede-project.org/index.php?do=details&task_id=292

danrl, 2016-11-19:
Is this a bug then in the dependency handling or in luci-proto-wireguard? Shouldn't it work with virtual packages as well?

Since it's not complicated magic we are doing here, I am fine with changing the dependency. I chose the virtual package intentionally, but don't mind changing it. Shall we?

hnyman, 2016-11-19:
> I don't see any reference to a wireguard package anywhere on the buildbots: it does not exist as a package and is not present in Packages.gz.

Hmmm. I see wireguard-tools and kmod-wireguard packages available for download, but not the wireguard package itself.

If I remember correctly, a package needs an "install" section for an .ipk file to be generated. wireguard virtual package seems to have no install section, so the .ipk does not get built and the dependency breaks there.

Examples (mainly various plugin/tool collections) of having a phony "true" install section:
https://github.com/openwrt/packages/blob/master/utils/coreutils/Makefile#L100
https://github.com/openwrt/packages/blob/master/utils/open-plc-utils/Makefile#L93
```
define Package/open-plc-utils/install
	true
endef
```
@jow closed LEDE issue 292, but I think that he overlooked the virtual package aspect.

My hunch is that the "correct" solution would be to add a similar "true" install section to wireguard. Then the ipk would be built and installing that would just call the dependent other packages.

Or do you @danrl have a wireguard*.ipk package in your /bin after compilation? I guess not.

danrl, 2016-11-19:
> Or do you @danrl have a wireguard*.ipk package in your /bin after compilation? I guess not.

No such package.

hnyman, 2016-11-19:
You might experiement a bit with wireguard. If you add the install section, does ipk get created? And does installing that ipk then install the two actual packages?

hnyman, 2016-11-19:
@zorun 
Is that meta-package / virtual package needed at all? 

Why not have just the userspace package to depend on the kernelpackage? That would simplify the dependencies and packages.

danrl, 2016-11-19:
> Why not have just the userspace package to depend on the kernelpackage? That would simplify the dependencies and packages.

Genereally, one might like to have `wg` without the kernel module. For example, I use `wg` on a Mac to generate keys for WireGuard VPNs, but there is no kernel module for macOS. Conceptually, I'd like to separate those two and not create a dependency there. It might not make such sense on embedded devices though. I really do love the packages to be separated and rather add multiple dependencies than squishing everything in one package.

zorun, 2016-11-19:
Agreed with @danrl .  Also, at some point, there will be other implementations of wireguard (userspace in Go, etc) that will also be controlled by `wg`.

I can build a dummy `wireguard` package with dependencies on `kmod-wireguard` and `wireguard-tools` (I thought it was already the case).  However, it would waste a few bytes on small devices for nothing. I think it's better if the luci package depends on `kmod-wireguard` and `wireguard-tools`.

hnyman, 2016-11-19:
ok, feel free to create PRs to fix things suitably.

But then the current metapackagecontinues to work only in menuconfig & .config, but it is invisible for people who install packages manually (from snapshot or release repo). They will need to install the two packages separately.

danrl, 2016-11-19:
PR #857 should do the trick.

PR #857 (2016-11-19): luci-proto-wireguard: do not depend on meta-package

Fixes #855
closed 2016-11-19 by hnyman