openwrt/luci #856: unsaved changes: 0

bol-van, 2016-11-19:
Whatever I change, save but not apply - subj is in the right-top corner.
Changes can be seen in /tmp/.uci
This bug lasts for years

cshoredaniel, 2016-12-09:
@bol-van Can you confirm the version of openwrt and LuCI for which this occurs?

cshoredaniel, 2016-12-09:
@bol-van Also what theme are you using?

bol-van, 2016-12-10:
git-16.306.57050-d7d9772-1
theme - openwrt
looks like bootstrap theme  does not have this indication
and thats why nobody cares because openwrt theme is abandoned

jow-, 2016-12-10:
I cannot reproduce that issue with either Bootstrap or OpenWrt themes.
[image]
[image]

cshoredaniel, 2016-12-10:
@bol-van Do you have any custom modifications?  Out of memory? What device? Are you using a snapshot? From where? Did you build your own?  From what repos, and what versions?

bol-van, 2016-12-10:
Custom modifications : no, i didnt modify luci
Out of memory : cant be. it happens even on 1gb ram device.
What device : wndr3800, wndr3700, dir825b2, banana pi. All either run DD or LEDE.
Own firmware : yes, i always build my own, never use snapshots.
What repos : standard git checkout of latest DD/lede

bol-van, 2016-12-10:
Experiment.
I setup LEDE in vmware.
Installed luci there. Works.
Copied /usr/lib/lua/luci from problematic system to vmware lede. Replaced 3 .so inside to x64 version.
It works !

cshoredaniel, 2016-12-10:
That's some sort of compilation issue then (unclean tree?)  Closing 'not a luci bug'

closed 2016-12-10 by cshoredaniel

bol-van, 2016-12-12:
I traced the reason to libuci.so.
(file.c) uci_list_config_files enumerates all files in /etc/config including directories
if directory is found then get_filename returns null
configs[i] remains unset and in most cases contain NULL
all part of code that use config enumeration stop on first NULL
If a directory exist in /etc/config then all fails

```
	char **configs = NULL;
	char **p;
	if ((uci_list_configs(ctx, &configs) != UCI_OK) || !configs) {
		cli_perror();
		goto out;
	}

	for (p = configs; *p; p++) {
		package_cmd(cmd, *p);
	}

```

bol-van, 2016-12-12:
This patch to file.c fixes the problem
```

838c838
<       int size, i;
---
>       int size, i, j;
862c862
<       for(i = 0; i < globbuf.gl_pathc; i++) {
---
>       for(i = 0, j = 0; i < globbuf.gl_pathc; i++) {
872c872
<               configs[i] = buf;
---
>               configs[j++] = buf;

```

jow-, 2016-12-12:
Thanks for figuring this out @bol-van . This has been recently patched upstream in uci.git, see http://git.lede-project.org/49ec6ef .

cshoredaniel, 2016-12-13:
Yes thank you @bol-van I obviously misdiagnosed; thanks @jow-