openwrt/luci #858: LuCI disconnects after logging in

CoderFX, 2016-11-20:
So I login to https://192.168.1.1, status window looks OK, then try going into any other tab, end up at login screen. This is default luci installation on openwrt.
Then I cannot use chrome to access my router. Firefox works fine for some time and this happens again. Now I end up using my last browser Safari to roll back any settings, but issue persists on Chrome and Firefox.

Chrome console log:
`xhr.js:76 GET https://192.168.1.1/cgi-bin/luci 403 (Forbidden)XHR.get @ xhr.js:76XHR._r @ xhr.js:186
;stok=b1f0db4…:117 Uncaught TypeError: Cannot read property 'wan' of null(…)(anonymous function) @ ;stok=b1f0db4…:117xhr.onreadystatechange @ xhr.js:72
/cgi-bin/luci:1 GET https://192.168.1.1/cgi-bin/luci 403 (Forbidden)
Navigated to https://192.168.1.1/cgi-bin/luci`

darkgnostic, 2016-11-30:
Same here, I login, do something, then I'm threw back at login screen, except that this happens after the install, somewhere after I add wireless interface. However, no error in console log. Using Chrome. 

@CoderFX thank you on Safari tip. At least it is working in Safari.

bananos, 2016-12-04:
This is an old issue.  The fix is to use **only one** tab of LuCI in your browser.

cshoredaniel, 2016-12-09:
@CoderFX @darkgnostic @bananos Can you confirm the following:

- Version of LuCI & openwrt on which this occurs
- Whether latest LuCI has the same issue

bananos, 2016-12-10:
@cshoredaniel I can confirm that I've seen this issue on 
 `OpenWrt Chaos Calmer 15.05.1 / LuCI 15.05-149-g0d8bbd2 Release (git-15.363.78009-956be55)` using Google Chrome (OSX 10.11.6)

However, issue seems to be resolved when testing on 
`Powered by LuCI Master (git-16.248.37139-a69bd42) / OpenWrt Designated Driver 49928`
using Firefox 50.0.2 (OSX 10.11.6)

cshoredaniel, 2016-12-10:
@bananos thank you for the info; I'm not sure it's feasible to fix for 15.05 without being too intrusive for a stable release; @jow- thoughts?

cshoredaniel, 2016-12-10:
This bug would be too intrusive to fix for CC and is fixed in trunk, hence closing WONTFIX.

closed 2016-12-10 by cshoredaniel

Deepak72840, 2017-01-24:
I am facing the same issue and I went through the posts on this topic and it seems the only fix for this issue is "we have to use only one tab". Is it the fix or the fix is something else???

darkgnostic, 2017-01-24:
I used only one tab, probably something was left in the cache. I have uninstalled luci, but Safari didn't made any problems if I remember well.

Deepak72840, 2017-01-24:
I also found some fix like below, however this fix is not applicable in case of me in dispatcher.lua file.
Can someone guide me on this???
modules/luci-mod-rpc/luasrc/controller/rpc.lua file 
-		require "luci.config"
+		local config = require "luci.config"
-			local sdat = util.ubus("session", "create", { timeout = luci.config.sauth.sessiontime })
+			local sdat = util.ubus("session", "create", { timeout = config.sauth.sessiontime })
Should I have to use only one tab or any other fix is also available.
These are the below luci versions I am using ;
admin@Cupola:/usr/lib/lua/luci# opkg list | grep luci
luci - git-17.017.42240-0c67964-1
luci-app-firewall - git-17.017.42240-0c67964-1
luci-base - git-17.017.42240-0c67964-1
luci-lib-ip - git-17.017.42240-0c67964-1
luci-lib-nixio - git-17.017.42240-0c67964-1
luci-mod-admin-full - git-17.017.42240-0c67964-1
luci-proto-ipv6 - git-17.017.42240-0c67964-1
luci-proto-ppp - git-17.017.42240-0c67964-1
luci-theme-bootstrap - git-17.017.42240-0c67964-1
admin@Cupola:/usr/lib/lua/luci#

Deepak72840, 2017-01-24:
Hi darkgnostic,
Thank you for your feedback.
Did you find any patch or solution for this issue???

Do you have any idea on how to set the session timeout value in luci???
Thanks in advance.

patricktokeeffe, 2018-01-12:
fwiw explicitly deleting the cookie isn't enough. I wound up searching, finding #392, then wading through my tabs to find the culprit. If my hardware supported trunk, I'd upgrade.

Virthuss, 2019-07-25:
Bug encountered with the version `git-18.205.33462-6bfc320`
Fixed by keeping only one LuCI tab opened.