openwrt/luci #867: VLAN order messed up on TL-WR842N-V2 and interface eth1.x won't come up.

rafwes, 2016-12-06:
Hello!

Yesterday I created an LEDE image with imgbuilder for my TP-Link WR842N-V2 and the port order is totally messed up.

Here is what I found out how the ports are mapped:

```
 physical       swconfig       luci (*)
=======================================
     0              0           LAN4
     1              2           LAN2
     2              3           LAN1
     3              4           CPU
     4              1           LAN3
=======================================

(*) LuCI Master (git-16.339.77170-5127dd6)
    LEDE Reboot CURRENT r2400+38-a245772
```

As you can see, it looks like LUCI somehow reversed swconfig's port order. 

Besides that I also noticed that creating a new vlan with VID 4 will fail to create a new interface eth1.4, I had to do that manually in the config files by adding `	option ifname 'eth1.4'` to a interface in `/etc/config/network` for it to show up

jow-, 2016-12-06:
The missing `eth1.4` is by design, the system will onyl create it once you use it somewhere. Merely declaring the swconfig vlan results in no Linux net device.

The port order you see in LuCI follows what LEDE specifies in the board defaults:
https://github.com/lede-project/source/blob/master/target/linux/ar71xx/base-files/etc/board.d/02_network#L281

The order is supposed to follow whats printed on the routers case on the outside. What would be the correct mapping of the external port numbers to the swconfig ones (ignoring LuCI) in your case?

rafwes, 2016-12-06:
> What would be the correct mapping of the external port numbers to the swconfig ones (ignoring LuCI) in your case?

This is shown in the table above. As an example on how to correctly read row4:
_If a PC is connected to the physical port 3 (as shown in the casing), swconfig will show port 4 as active and LUCI will show CPU as active._

Beware, I have here some routers from TPLink and even between versions of the same model, swconfig's response under openwrt was different regarding active ports. So, this change should only be apply to WR842N-V2 and NOT V1 or V3, since I cannot tell how swconfig behaves for them.

Just to clarify what's going on with eth1.4. On openwrt I would create a new VLAN and immediately after that a new interface would appear as an option under [physical settings](https://cloud.githubusercontent.com/assets/23530231/20930481/a272766c-bbb4-11e6-8c44-ab14aff99772.png). This is no longer the case, the option "Software VLAN: eth1.4" simply wasn't there to be selected. There was therefore no way to "use" the new interface from LUCI, I had to assign it manually to the bridge.

jow-, 2016-12-06:
I was mainly wondering about the fact that swconfig responds to all five external ports because the blue wan one is supposed to be on a different nic (eth0).

So from the info in your table combined with a photo I found online (http://biblebox.org/wp-content/uploads/2014/08/IMG_0270-1024x768.jpg), the schema would be:

            Power  LAN4    LAN3    LAN2    LAN1    WAN     USB
             (*) | [...] | [...] | [...] | [...] | [...] | [.] 
    swconfig port:   1       4       3       2      n/a
    netdev:        eth1    eth1    eth1    eth1    eth0
    
    swconfig port 5: unconnected
    swconfig port 0: CPU (connection to eth1)

Is the above diagram correct?

jow-, 2016-12-06:
Btw, can you also please provide a copy of your current `/etc/board.json` ?

rafwes, 2016-12-06:
You diagram is correct. There is no Port5 in swconfig. As you correctly guessed the blue (wan) port is another physical interface (eth0), completely detached from the switch.

As you asked: [board.json](https://github.com/openwrt/luci/files/634339/board.json.txt)

jow-, 2016-12-06:
Okay, thank you. I think the missing `eth1.4` is a consequence of them isdetected CPU port (LuCI only offers an `ethX.VID` if what it considers to be the CPU port is tagged.

jow-, 2016-12-06:
Can you please try the following fix: http://git.lede-project.org/e19b661

You can apply it manually to your device by editing `/etc/board.d/02_network`, then delete your `/etc/board.json` and `/etc/config/network` and finally run `/bin/board_detect`.

This will rebuild `board.json` to reflect the updated switch layout and should cause LuCI to properly display the switch topology.

rafwes, 2016-12-06:
Did as you told me but it made matters worse. 

There is a new VLAN with VID2 created by default on port 4 and 0. Normally this only happens for devices where WAN resides on the same switch, like WDR4300-V1.

Apart from that, it creates now twice as many ports. This is the new [board.json](https://github.com/openwrt/luci/files/634739/board.json.txt) and can see [8 ports](https://cloud.githubusercontent.com/assets/23530231/20939123/ba249662-bbd4-11e6-85f1-bdeb637967ba.png) in LUCI. The new `/etc/config/network` reads:

```
config switch_vlan
	option device 'switch0'
	option vlan '1'
	option ports '1 2 3 4 0 1 2 3 4'

config switch_vlan
	option device 'switch0'
	option vlan '2'
	option ports '0 4'
```

rafwes, 2016-12-06:
I might have fucked up something. I tried reverting /etc/board.d/02_network to a saved version but I keep getting 8 ports. Hold on, I'll try again.

rafwes, 2016-12-06:
Ok, this is getting even stranger. I thought I had made an error by editing 02_network and downloaded [this version from git](https://github.com/lede-project/source/blob/master/target/linux/ar71xx/base-files/etc/board.d/02_network) and copied it to /etc/board.d/ in order to restore everthing. Afterwards I deleted board.json and rebooted the router.

Now I have 12 ports on the VLAN interface. No idea what's causing this.

cshoredaniel, 2017-01-17:
@rafwes if you do a fresh install do you still have this issue?

rafwes, 2017-01-24:
@cshoredaniel: I saw the patch has been integrated into LEDE. As soon as we get a release candidate of 17.01 with a clear upgrade path the final release, I will test it. I'll come back to you then.

rafwes, 2017-02-04:
@cshoredaniel @jow-

I've just flashed the router with 17.01-rc1 and can confirm that the port order is correct and the interface eth1.4 shows up as expected. Kudos and thanks for that! However I had a soft-brick when creating a new VLAN with LUCI, requiring to boot the device into failsafe to recover. Here is what happens:

By default, VLAN is not turned on the device's switch, since WAN resides on another physical interface. So `br-lan` bridges `wifi0` and `eth1`. If you activate VLAN, all ports are assigned to VID1 but `/etc/config/network` is not updated accordingly to bridge `eth1.1` instead of `eth1`, rendering the device inaccessible.

One solution would be to turn VLAN by default on the factory images or take care of `/etc/config/network` when activating VLAN for the first time.

Again, thanks for all the help, you're amazing.

cshoredaniel, 2017-02-07:
@rafwes Thanks for the report!  You might want to discuss the vlan issue with upstream; I've noticed the same behaviour if one uses uci (not luci) to enable the switch but forget to enable the vlan.  It probably make sense to open that as a new 'feature request' issue.

closed 2017-02-07 by cshoredaniel