openwrt/luci #868: problem with delete interface

palat, 2016-12-09:
HI 

i've build opewrt and luci with src from your repo everything work perfectly but when i delete the interface
just got this message on my browser

it happend after apply configuration and jump back to network page

please see messgage below

how can i fix this problem

No page is registered at '/admin/network/network'.
If this url belongs to an extension, make sure it is properly installed.
If the extension was recently installed, try removing the /tmp/luci-indexcache file

thank you and sorry for my poor english

maverix

hnyman, 2016-12-09:
works for me.
Please explain more closely the details, which Openwrt version, how/where you are deleting the interface, what kind of interface you are deleting, etc.

(When I delete an interface from the Network/Interfaces overview page, there is no need to separately "apply configuration")

jow-, 2017-01-04:
No response after a month

closed 2017-01-04 by jow-