openwrt/luci #869: [RFC]: Decomposing modules (preparation for shift to LuCI2)

cshoredaniel, 2016-12-09:
Hi all,

I'm feeling better and am thinking that I'd like to be useful by starting to take steps to make a smooth transition from current LuCI (Lua+HTML+a little javascript) to LuCI2 (ubus rpc + HTML5 + lots of javascript).

I think the first step is to decompose some of the major components like mod-admin-full into applications so that individual pieces can be replaced more easily, instead of having a situation where the all of mod-admin-full has to be completely rebuilt from the ground up for the replacement to be done.

So my plan is to identify common components and moving them into 'library' packages, and to move admin-full pages into individual applications under the applications directory.

As I move components out of mod-admin-full, I intend to make mod-admin-full depend on the new applications, so that from the user perspective nothing changes for now.

Any thoughts?

jow-, 2016-12-09:
@cshoredaniel - sounds good to me! I've noticed that there's a bit of confusion around "luci2" so I'd like to take the opportunity to clarify what "luci2" is going to be.

There's two repositories:
1) the "luci2/ui.git" repository (https://git.lede-project.org/?p=project/luci2/ui.git) which was a first prototype to build client side luci using a custom framework
2) the "luci-ng" repository on Github (https://github.com/jow-/luci-ng/)

To not fall into the custom code/framework trap again like we did with LuCI, I'd like to consolidate future work in jow-/luci-ng based on Angular.js to base our work on a well known, maintained foundation.

I will add a deprecation notice to the other repository to avoid future misunderstandings.

cshoredaniel, 2016-12-09:
@jow- Ok, thank you for the clarification; I didn't know about that.  Angular.js sounds like a good plan to me.  I've taken a quick pick at your luci-ng repo and it looks like like a plan I have ought to be doable (which is to have a luci-ng compatibility shim that allows using luci-ng alongside current luci with the same basic look, although the feel will be probably be different because of more responsive ui from luci-ng); basically my goal, long term, is to make it possible to shift individual apps (including decomposed mod-admin-full) without a lot of fuss, until enough has been shifted to deprecate luci and put it in maintenance mode and stop accepting new apps for the Lua stuff.

cshoredaniel, 2016-12-09:
Note, that I won't push the work in this regard until after the next release, unless that is being pushed significantly out due to possible merger.

cshoredaniel, 2017-07-12:
For personal reason no longer participating in OpenWrt/LEDE.

cshoredaniel, 2018-08-01:
Reopening since it's back on my radar.

cshoredaniel, 2018-08-01:
@jow made an important point here https://github.com/openwrt/luci/issues/540#issuecomment-409464192; too many packages results in packaging related bloat.

cshoredaniel, 2018-08-01:
@jow- How would the bloat be for a generic node/entry enable/disable switch (so unwanted nodes/entries don't show up in the luci menus)?  Does that sound like a reasonable alternative?

cshoredaniel, 2018-08-01:
Started some discussion on the forum here: https://forum.lede-project.org/t/luci-coding-making-sure-higher-level-menus-exists-from-a-leaf-page/17929/2?u=cshoredaniel

cshoredaniel, 2018-08-01:
@hnyman asks what is the benefit: here are my reasons:

    To make it easier to pick and choose what ot use when doing a combination of luci and luci-ng (or third party luci replacements beings worked on), so that having a functional gui doesn't depend on building 'all' of core modules before it's a viable gui.

    For things like @EricLuehrsen replacing dnsmasq, or if someone wants to use unbound + a (yet to be done) isc-dhcp-server app that it's not blocked by core stuff.

    I personally prefer granuality to monolithic.

The packaging issue does, however, complicate the question a little.

jow-, 2018-08-01:
One thing I like to be implemented as a first step is making "luci-base" working standalone. Means solely installing `luci-base` will provide the ability to login, logout, review uci changes and apply/revert them.

In addition, it should provide a placeholder page for `/admin` printing something like `No component installed`.

As a next step we could think about splitting luci-mod-admin-full into the three subcategories `status`, `network` and `system`.

Once we reached that milestone we can see if it makes sense to further break down components.

cshoredaniel, 2018-08-01:
Sounds like a plan to me.  @hnyman any objections?