openwrt/luci #875: Wifi.lua - Probe EAP support

KarHoey, 2016-12-12:
In luci/model/cbi/admin_network/wifi.lua

`-- Probe EAP support`
`local has_ap_eap  = (os.execute("hostapd -veap >/dev/null 2>/dev/null") == 0)`
`local has_sta_eap = (os.execute("wpa_supplicant -veap >/dev/null 2>/dev/null") == 0)`

`if hostapd and supplicant then`
`...`
`                       if has_ap_eap and has_sta_eap then`
                        `encr:value("wpa", "WPA-EAP", {mode="ap"}, {mode="sta"}, {mode="ap-wds"}, {mode="sta-wds"})`
                        `encr:value("wpa2", "WPA2-EAP", {mode="ap"}, {mode="sta"}, {mode="ap-wds"}, {mode="sta-wds"})`
                 `end`

root@OpenWrt:~# hostapd -veap
hostapd v2.5
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2015, Jouni Malinen <j@w1.fi> and contributors

root@OpenWrt:~# wpa_supplicant -veap
wpa_supplicant v2.5
Copyright (c) 2003-2015, Jouni Malinen <j@w1.fi> and contributors

I found that the output of "wpa_supplicant -veap" is considered as STDOUT, but "hostapd -veap" is considered as STDERR, hence it could never turn on EAP options in LUCI Wi-Fi.

Should we modify the has_ap_eap condition so that
`-- Probe EAP support`
       ` local has_ap_eap  = (os.execute("hostapd -veap 2>/dev/null >/dev/null") == 0)`
       ` local has_sta_eap = (os.execute("wpa_supplicant -veap >/dev/null 2>/dev/null") == 0)`

Thank you.

jow-, 2016-12-12:
Its not about STDOUT or STDERR but about the exit code of the command. There's an optional patch in OpenWrt and LEDE which is supposed to do that:
https://github.com/openwrt/openwrt/blob/master/package/network/services/hostapd/patches/420-indicate-features.patch

KarHoey, 2016-12-14:
Dear Jow,
Thank you for the reply. I got it :+1:

closed 2016-12-15 by KarHoey