openwrt/luci #882: IPv6 Unable to set up router in relay mode for IPv6 addresses over DHCPv6

JKAbrams, 2016-12-16:
This is a feature request.

Here is the config I want to archive. I have marked how the defaults (on ***Designated Driver***) needs to be changed for my setup to work with my ISP (having my clients behind my router receiving global IPv6 addresses from my ISPs DHCPv6):

/etc/config/network
```Bash
config dhcp 'lan'
        option interface 'lan'
        option start '100'
        option limit '150'
        option leasetime '12h'
        option ra 'relay'       # Changed to relay
        option dhcpv6 'relay'   # Changed to relay
        option ndp 'relay'      # Added

# Section commented
#config dhcp 'wan'
#       option interface 'wan'
#       option ignore '1'

# Section added
config dhcp 'wan6'
        option interface 'wan6'
        option dhcpv6 'relay'
        option ra 'relay'
        option ndp 'relay'
        option master '1'
```

**Interfaces - LAN - [Edit] - DHCP Server - IPv6 Settings** takes care of the first section by setting **Router Advertisement-Service**, **DHCPv6-Service** and **NDP-Proxy** to **relay mode**.

[image]

**But (as far as I know) there is no way to:**
1. Get rid of the dhcp 'wan' section and (*not sure if this is needed*)
2. Add the dhcp 'wan6' section including the relay options (this is needed for this setup to work)

jow-, 2016-12-16:
1. There is no need to get rid of the `wan` section, simply set it to ignore (it should be ignored by default already).
2. Yes, this might indeed be problematic - I guess if you navigate to the WAN6 interface you're not being offered an option to setup the DHCP server?

JKAbrams, 2016-12-16:
1. Thanks for the comment, I'll try setting it to ignore and see if that works.
2. Correct, the DHCP settings does not appear for WAN6.

JKAbrams, 2016-12-16:
Regarding 1. I just tested leaving the wan section as is with the ignore setting set to 1 and I can confirm that works with my setup.

JKAbrams, 2016-12-18:
Sorry, I was too quick in my last comment.
The **'wan'** section cannot be present for my setup to work.

Rebooting the services:
**root@OpenWrt:/# /etc/init.d/./dnsmasq restart
root@OpenWrt:/# /etc/init.d/./odhcpd  restart**
was not enough for the **/etc/config/dhcp** file to get reread, to test changes I needed to reboot the router.

JKAbrams, 2016-12-18:
*This is only slightly related, but I put this here for completeness.*

After upgrading these packages (on Designated Driver):

luci-lib-ip - git-16.341.34736-6d5f878-1 - git-16.349.29317-60bb727-1
luci-theme-bootstrap - git-16.341.34736-6d5f878-1 - git-16.349.29317-60bb727-1
base-files - 169-50011 - 169-50013
luci-app-firewall - git-16.341.34736-6d5f878-1 - git-16.349.29317-60bb727-1
luci-proto-ppp - git-16.341.34736-6d5f878-1 - git-16.349.29317-60bb727-1
luci-mod-admin-full - git-16.341.34736-6d5f878-1 - git-16.349.29317-60bb727-1
luci-base - git-16.341.34736-6d5f878-1 - git-16.349.29317-60bb727-1
luci-proto-ipv6 - git-16.341.34736-6d5f878-1 - git-16.349.29317-60bb727-1
luci-lib-nixio - git-16.341.34736-6d5f878-1 - git-16.349.29317-60bb727-1
luci-lib-jsonc - git-16.341.34736-6d5f878-1 - git-16.349.29317-60bb727-1
luci - git-16.341.34736-6d5f878-1 - git-16.349.29317-60bb727-1

pinging IPv6 on the client side stopped working. If the 'wan' section is gone IPv6 assignments works, but no IPv6 traffic gets through the router. It is not a DNS issue as pinging IPv6 IP directly does not work.

ianzhou233, 2016-12-18:
I used native IPV6 in my school,and the router get IPV6 address,but no ipv6 address on my phone and laptop(wired connected). Finally i installed 6relayd.ipk to make my wireless worked with IPV6,and that works.It is a package abandoned over openwrt 13.
BTW,I added 6relayd's to /etc/rc.d to make it work with start.
hope I can help you ^_^

bastien-roucaries, 2017-01-08:
Hi

Something like this is working for me I have decomposed the lan in 4 interfaces. One for dhcp, one for ipv4, two for ipv6 (private + public)

/etc/config/network:
config interface 'A_MAIN'
        option ifname '@BRIDGE_V20'
        option _orig_ifname '@BRIDGE_V20'
        option _orig_bridge 'false'
        option proto 'static'

config interface 'LAN_V20'
        option proto 'static'
        option ifname '@BRIDGE_V20'
        option netmask '255.255.255.0'
        option ipaddr '192.168.20.1'

config interface 'LAN_V20_6'
        option ifname '@BRIDGE_V20'
        option _orig_ifname '@BRIDGE_V20'
        option _orig_bridge 'false'
        option proto 'static'
        option ip6addr '2a01:e35:2f01:a61::1/64'

config interface 'LAN_V20_6L'
        option proto 'static'
        option ifname '@BRIDGE_V20'
        option ip6addr 'fd02:15b3:9237:20::1/64'

/etc/config/dhcp:

config dhcp 'A_MAIN'
        option start '100'
        option leasetime '12h'
        option limit '150'
        option interface 'A_MAIN'
        option ra 'server'
        option dhcpv6 'server'
        option ra_management '1'
        option ra_default '1'

config dhcp 'LAN_V20'
        option interface 'LAN_V20'
        option start '100'
        option limit '150'
        option leasetime '12h'
        option dhcpv6 'disable'
        option ra 'disable'
        option ndp 'disable'

config dhcp 'LAN_V20_6'
        option interface 'LAN_V20_6'
        option ra 'server'
        option dhcpv4 'disable'
        option dhcpv6 'server'
        option ra_management '1'
        option ra_default '1'

config dhcp 'LAN_V20_6L'
        option interface 'LAN_V20_6L'
        option ra 'server'
        option dhcpv6 'server'
        option ra_management '1'
        option ra_default '1'
        option dhcpv4 'disable'

cshoredaniel, 2017-01-16:
@JKAbrams does something like @bastien-roucaries solution work for you?

JKAbrams, 2017-01-22:
@cshoredaniel sorry, for late reply, I've not tried it but I could if you think it would help solving the problem, to me it looks like a rather complicated setup to solve a simple (?) problem. I'm not sure I understand what problem decomposing into different VLANs is supposed to solve or how it relates to the fact that I'm unable to use Luci to configure my router into a working configuration, this bug report is meant toward finding a solution to that problem.

**To recap:**
I try to set my network up in the most common sense way I can think of that actually works with my ISP, please let me know if you I'm somehow mistaken here as my knowledge of networking is limited.
Since IPv6 addresses are globally unique to me it makes sense to have each client have their own so they are addressable therefore the router should let these assignments from my ISP's DHCP through, but since most of the world is still IPv4 the router still needs to be configured for IPv4 NAT.
With a default config, this does not work for me, I've failed to find a way to use the Luci interface to archive this configuration, I did however find a way to configure the router manually as documented above that worked (until it broke by the update, but I think this is most likely due to an unrelated bug as the config above is rather straight forward and based entirely on the information I could find on the IPv6 guide on openwrt.org).