openwrt/luci #885: luci-app-statistics shows "0" or "-255" for "Noise" value on WRT1900ACSv1 with Kernel 4.4.38 and latest driver

cybrnook, 2016-12-17:
luci-app-statistics shows "0" or "-255" for "Noise" value on WRT1900ACSv1 with Kernel 4.4.38 and latest driver 10.3.2.0-20161124.

http://img.techpowerup.org/161217/0.png

and 

http://img.techpowerup.org/161217/255.png

cybrnook, 2016-12-17:
Also from davids thread, post 744:
https://forum.openwrt.org/viewtopic.php?id=64949&p=30

"So it must be Kernel 4.4.36 (and newer) with 1124 driver (and newer) = 0dBm noise reported by luci-app-statistics."

cybrnook, 2016-12-17:
Adding @hnyman who I believe is the maintainer of this.

hnyman, 2016-12-17:
Well, LuCI statistics only shows graphs based on data collected by collectd daemon (from packages feed) and the wifi stats in collectd rely on an iwinfo addition by Openwrt that fetshed data from the driver. Likely something has changed either in the iwlwifi driver (so that has stopped it providing data) or in iwinfo reading the data for collectd. I am not really familiar with the iwinfo part, but @jow- might know more.

But if iwlwifi driver 10.3.2.0-20161011 provides data and 10.3.2.0-20161124 does not, the reason may quite well be in the driver itself.

(And I am not aware of any dependence on snmp, so that part in the forum thread looks strange.)

hnyman, 2016-12-18:
@cybrnook 
Does iwinfo work normally from console? It should provide you an overview to the wifi data, including signal and noise levels. That is the data polled by collectd. If you see no values on console with ` iwinfo `, there is not much either collectd or luci statistics can do.
```
root@lede:~# iwinfo
wlan0     ESSID: "XXXWLAN"
          Access Point: DC:EF:..:..:..:E6
          Mode: Master  Channel: 36 (5.180 GHz)
          Tx-Power: 23 dBm  Link Quality: 67/70
          Signal: -43 dBm  Noise: -108 dBm
          Bit Rate: 6.0 MBit/s
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11nac
          Hardware: 168C:0046 168C:CAFE [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy0

wlan1     ESSID: "ZZZWLAN"
          Access Point: DC:EF:..:..:..:E7
          Mode: Master  Channel: 3 (2.422 GHz)
          Tx-Power: 20 dBm  Link Quality: 70/70
          Signal: -27 dBm  Noise: -97 dBm
          Bit Rate: 1.0 MBit/s
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: 168C:0046 168C:CAFE [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy1
```

cybrnook, 2016-12-18:
Testing against Kernel 4.4.38 with driver 10.3.2.0-20161011 on WRT1900ACSv1, the graphs work properly. However I cannot execute iwinfo from console. It returns the following:

root@LEDE-WRT1900ACS:~# iwinfo
-ash: iwinfo: not found
root@LEDE-WRT1900ACS:~# which iwinfo
root@LEDE-WRT1900ACS:~# echo $PATH
/usr/sbin:/usr/bin:/sbin:/bin
root@LEDE-WRT1900ACS:~#

cybrnook, 2016-12-18:
root@lede:# uname -a
Linux lede 4.4.38 #0 SMP Sat Dec 17 21:23:49 2016 armv7l GNU/Linux
root@lede:# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info | grep driver
driver name: mwlwifi
driver version: 10.3.2.0-20161124
root@lede:# iwinfo
wlan0     ESSID: "GENERIC50GHZ"
          Access Point: XX:XX:XX:XX:FF:40
          Mode: Master  Channel: 149 (5.745 GHz)
          Tx-Power: 23 dBm  Link Quality: 41/70
          Signal: -69 dBm  Noise: unknown
          Bit Rate: 617.5 MBit/s
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11nac
          Hardware: 11AB:2A55 11AB:0000 [Marvell 88W8864]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

wlan1     ESSID: "GENERIC24GHZ"
          Access Point: XX:XX:XX:XX:FF:3F
          Mode: Master  Channel: 3 (2.422 GHz)
          Tx-Power: 30 dBm  Link Quality: 70/70
          Signal: -40 dBm  Noise: unknown
          Bit Rate: 90.2 MBit/s
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: 11AB:2A55 11AB:0000 [Marvell 88W8864]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1

root@lede:#

cybrnook, 2016-12-18:
and on older driver:

root@LEDE-WRT1900ACS:# uname -a
Linux LEDE-WRT1900ACS 4.4.38 #0 SMP Sat Dec 17 21:23:49 2016 armv7l GNU/Linux
root@LEDE-WRT1900ACS:# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info | grep driver
driver name: mwlwifi
driver version: 10.3.2.0-20161011
root@LEDE-WRT1900ACS:# iwinfo
wlan0     ESSID: "WIFI_5000MHZ"
          Access Point: 16:91:82:BB:FF:40
          Mode: Master  Channel: 149 (5.745 GHz)
          Tx-Power: 23 dBm  Link Quality: 47/70
          Signal: -63 dBm  Noise: -87 dBm
          Bit Rate: 564.9 MBit/s
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11nac
          Hardware: 11AB:2A55 11AB:0000 [Marvell 88W8864]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

wlan1     ESSID: "WIFI_2400MHZ"
          Access Point: 16:91:82:BB:FF:3F
          Mode: Master  Channel: 3 (2.422 GHz)
          Tx-Power: 30 dBm  Link Quality: 68/70
          Signal: -42 dBm  Noise: -84 dBm
          Bit Rate: 72.2 MBit/s
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: 11AB:2A55 11AB:0000 [Marvell 88W8864]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1

cybrnook, 2016-12-18:
So you believe this to be a bug with the driver, and not statistics/collectd?

hnyman, 2016-12-18:
Yeah, likely nothing to be node at Luci, and also nothing to be done with collectd.

closed 2016-12-18 by hnyman

cybrnook, 2016-12-19:
Thanks @hnyman, I have opened a report in the MWLWIFI git.

https://github.com/kaloz/mwlwifi/issues/129