openwrt/luci #887: luci-app-openvpn - VPN Tab Settings Missing

Presjar, 2016-12-18:
When using luci-> Services -> OpenVPN -> Switch to advanced configuration -> Tab 'VPN' 
On click, the window shows all the expected settings for a split second and then changes to show only 'client' tickbox Configure client mode.

luci-app-openvpn git-16.352.65143-1741e55-1
LEDE Reboot SNAPSHOT r2524-aada15a / LuCI Master (git-16.352.65143-1741e55)

paulbalomiri, 2017-01-03:
Hello, i can confirm the same problem (with briefly appearing option ) in luci-app-openvpn - git-16.018.33482-3201903-1

Wedmer, 2017-04-25:
The problem is in filter options. Filter wants variable server_mode="1" to show those options, but there're no such variable. Later there will be fix for this.

commit f45d8767 (2017-04-25): [luci-app-openvpn] Fixed ivalid checks for parameters in VPN tab.
closes #887
closed 2017-04-25 by hnyman