openwrt/luci #913: compile error

callnothing, 2017-01-02:
when i git clone luci and checkout to luci-0.12, then run make
Error occurred

compile err lines：
svn: E000111: Unable to connect to a repository at URL 'svn://svn.openwrt.org/openwrt/branches/backfire/package/uhttpd/src'

hnyman, 2017-01-02:
You have likely stumbled upon this line:
http://git.openwrt.org/?p=project/luci.git;a=blob;f=contrib/uhttpd/Makefile;h=930bf9ae6a64011e2541ff903c0c60f94c62811c;hb=refs/heads/luci-0.10#l4

You should not be doing a make in the luci directory itself. Luci sources are supposed to be a feed in the Openwrt build system, and the needed modules will get built along all other Openwrt modules.

uhttpd has been part of the main Openwrt sources since 2012, so that Makefile you found is totally outdated (and has been removed in 2015).

jow-, 2017-01-04:
LuCI 0.12 is not supported anymore and the referenced repository has been removed.

closed 2017-01-04 by jow-