openwrt/luci #915: Allow to set dhcpv4 disabled

bastien-roucaries, 2017-01-02:
Hi,

I wish to disable dhcpv4 on some aliased (using @syntax) interface. Could be possible to add a check box and in this case disable the ipv4 panel

jow-, 2017-01-04:
Such an option already exists:
[image]

closed 2017-01-04 by jow-

bastien-roucaries, 2017-01-04:
No this set option ignore '1' that disable dhcpv6 and dhcpv4. I want to
disable only dhcp for v4 and use ra for v6.

Bastien

jow-, 2017-01-04:
Option ignore currently only disables dhcpv4 (assuming a standard setup with dnsmasq + odhcpd).

bastien-roucaries, 2017-01-04:
> Option ignore currently only disables dhcpv4 (assuming a standard setup
> with dnsmasq + odhcpd).

​Thanks documentation is really bad in this case...​