openwrt/luci #918: [openwrt/luci] ifaces.lua - HTML template to create DynamicList custom fields

KarHoey, 2017-01-04:
In LUCI Interfaces-LAN > DHCP Server > Advanced Settings > DHCP-Options, you have to type for example '3,192.168.2.1', then click on [+], then type '6,8.8.8.8'.

I want to do simplest GUI settings like

Alternative DNS : 
Alternative Gateway : 

I written the following codes 
### In luci/model/cbi/admin_network/ifaces.lua
dhcpopt3 = s:taboption("advanced", **DynamicList**, "dhcp_option", translate("Alternative Gateway"), translate(""))
dhcpopt3.template = "cbi_defrouter"
               function dhcpopt3.validate(self, value)
                        local defrouter = self.map:formvalue(self:cbid(section) .. ".defrouter")
                                return "3," .. defrouter    
                end

### In luci/view/cbi_defrouter.htm
<%+cbi/valueheader%>

<input type="text" class="cbi-input-text" onchange="cbi_d_update(this.id)"<%=attr("name", cbid .. ".defrouter") .. attr("id", cbid ..".defrouter") .. attr("value",
(self:cfgvalue(section) or ""):match("%d,(%d+.%d+.%d+.%d+)")) %> />

<%+cbi/valuefooter%>

It could not work. Unless I change the **DynamicList** to **Value** in luci/model/cbi/admin_network/ifaces.lua, then it can work and successfully save the IP into DHCP config file with DHCP option enum at the beginning, as shown below.

### In /etc/config/dhcp
config dhcp 'lan'
	...
	option dhcp_option '3,60.60.60.2'

It's not flexible, because "option dhcp_option" limits me to setup multiple DHCP option at one interface (the correct config is list dhcp_option '3,60.60.60.2 6,70.70.70.3'). What can I modify so that we can write template to create custom dynamic list field in LUCI view?

Thank you.

jow-, 2017-01-04:
You cannot express such a logic in CBI / HTML templates. You need to override the .write, .remove and .cfgvalue functions to decompose and join the values as needed.

When you have multiple fields presenting (parts of) the value of a single uci option then the usual approach is to let the `.write()` and `.remove()` function of the first field handle all the logic while the functions of the other fields are no-ops that do not actually write into uci.

To see a real worl example, refer to https://github.com/openwrt/luci/blob/master/protocols/luci-proto-ppp/luasrc/model/cbi/admin_network/proto_pppoe.lua#L73 and see how `keepalive_failure` and `keepalive_interval` write into the same `option keepalive` which takes two space separated numbers as value.

jow-, 2017-01-04:
When considering the example I linked, you need to keep in mind that you need to pass an array to the uci `set()` function in order to store the value as `list dhcp_option` instead of `option dhcp_option`.

Also keep in mind that the `get()` function can return either an array if the user already has `list dhcp_option ...` in his config or a string if the config contains `option dhcp_option ...` so within the `.cfgvalue()` callback you need to silently convert a string value to a one-element array if needed.

closed 2017-01-04 by jow-

KarHoey, 2017-01-09:
Thanks Jow for your tips, it helps me a lot. This is what I did.

-- create seperate field for user input DHCP option 3 (Router) and 6 (DNS)

		local dhcpopt6 = s:taboption("advanced", DynamicList, "dhcp_option", translate("Alternative Default DNS"), translate(""))
		dhcpopt6.datatype = "ip4addr"

		-- ip4addr datatype does not seems work in dhcpopt3
		local dhcpopt3 = s:taboption("advanced", DynamicList, "dhcp_option", translate("Alternative Default Gateway"), translate(""))
		dhcpopt3.datatype = "ip4addr"
		dhcpopt3.default = "0.0.0.0"

		--check dhcp option enum for dhcpopt3 and dhcpopt6 cfgvalue
		function dhcpopt6.cfgvalue(self, section)
			local v = m2.uci:get("dhcp", section, "dhcp_option")
			-- convert the string value to a one-element array
			local a = {}
			local op = 0
			local j = 1
			if v and #v > 0 then

				for i=1,#v,1 do
					-- discard one or more characters after the the comma
					op = v[i]:match("[^,]+")
					if op == "6" then
						-- print one or more characters after the the comma
						a[j] = v[i]:match("[,](.*)")
						j = j + 1

					end
				end

				return a
			end
		end

		function dhcpopt3.cfgvalue(self, section)
			local v = m2.uci:get("dhcp", section, "dhcp_option")
			local a = {}
			local op = 0
			if v and #v > 0 then

				for i=1,#v,1 do
					op = v[i]:match("[^,]+")
					if op == "3" then
						a[1] = v[i]:match("[,](.*)")
					end
				end

				return a
			end
		end

I wrote `.cfgvalue()` function to read IP from dhcp_option, and sort them out to respectively fields based on their dhcp option enum.

KarHoey, 2017-01-09:
However I'm lost in writing .write() function because either

m2.uci:get("dhcp", section, "dhcp_option") or 0 OR
dhcp_option:formvalue(section) or 0

return only IP address, without the dhcp option enum at the beginning, so I cannot differentiate whether the IP address is belong to dhcpopt3 or dhcpopt6.

I tried writing separate dhcpopt6.write() and dhcpopt3.write() but it don't work as well.

How can I retrieve value from uci files in lua? Thank you.