openwrt/luci #922: Not working with lighttpd anymore

alzhao, 2017-01-07:
I am using LEDE trunk. Luci works with uhttpd without problem. But when configured using Lighttpd, it doesn't work. It used to work in OpenWrt CC1505.

I can see the login interface, but there is no way to login. It always stays in the login page.
If I start using lighttpd then everything works. After I login, I switch back to lighttpd and I can using Luci without problem. so the problem only happens with login. Is this related to token or something?

gstrauss, 2017-01-07:
In lighttpd 1.4.42, lighttpd implemented
  RFC3875 CGI 1.1 specification section 6.2.2 Local Redirect Response
  http://www.ietf.org/rfc/rfc3875
for https://redmine.lighttpd.net/issues/2108

However, LuCI sends 302 Found redirect with Location: containing a URL path instead of a fully qualified URI.

See lighttpd patch in https://github.com/CZ-NIC/turris-os-packages/pull/20 which will be in lighttpd 1.4.45 (not yet released).

Another solution would be for LuCI to send back a fully qualified URI.  lighttpd puts the original request URi in the REQUEST_URI environment variable in CGI environments, which can be used to construct the URI.

gstrauss, 2017-01-07:
A possible fix to LuCI is to edit /usr/lib/lua/luci/http.lua:
```
--- http.lua 	2017-01-07 11:12:23.715932000 -0500
+++ http.lua.new	2017-01-07 11:10:51.505932000 -0500
@@ -241,7 +241,8 @@
 function redirect(url)
 	if url == "" then url = "/" end
 	status(302, "Found")
-	header("Location", url)
+	local scheme = getenv("HTTPS") and "https" or "http"
+	header("Location", scheme .. "://" .. getenv("SERVER_NAME") .. url)
 	close()
 end
 
```
However, for it to take effect you need to `rm -rf /tmp/luci-modulecache/`

alzhao, 2017-01-09:
Thanks. This works!!

closed 2017-01-09 by alzhao

gstrauss, 2017-01-09:
@alzhao, I am glad that works for you.  However, instead of closing this issue, shouldn't LuCI developers have a chance to evaluate if the change should be applied to LuCI?

gstrauss, 2017-01-09:
(https://github.com/openwrt/luci/blob/master/modules/luci-base/luasrc/http.lua)

laoshaw, 2018-02-23:
@gstrauss tried changing http.lua and used the 4 patches from 17.01.4 still the luci will show 403-forbidden.

also tried lighttpd 1.4.48, same problem. checking at 1.4.48 code it looks some of the local-redir patch are included already.

not working...

gstrauss, 2018-02-24:
@laoshaw my patches worked on the systems I tested over a year ago.  I suggested changes to opkg, which were incorporated into LEDE, but not into the then-current version of openwrt.  I suggested changes to the luci package, but those were not accepted.  I am neither an opkg nor a luci developer, and I do not maintain either package in openwrt.  Please don't post in such an old ticket and tag the wrong developers.  Open a new issue.

laoshaw, 2018-02-24:
will open a new ticket, thanks!