openwrt/luci #924: Could not add source nat on luci

bastien-roucaries, 2017-01-08:
When I click to add it  do:
firewall.cfg353837=redirect
firewall.cfg353837.enabled=1

But it does not show in snat section of luci...

bastien-roucaries, 2017-01-08:
I use git-17.007.54272-95d3f26

cshoredaniel, 2017-01-10:
Did you remember to actually defines the IPs etc (it looks like you added the new section but didn't populate it).  Can you do a screenshot?

HyP3r-, 2017-11-08:
I have the same error. I use git-17.290.79498-d3f0685.
Here are some screenshots:
[image]
[image]
[image]
[image]

systemcrash, 2018-01-23:
Same for me - I run

Powered by LuCI lede-17.01 branch (git-17.298.45605-07e7ff0) / LEDE Reboot 17.01.4 r3560-79f57e422d

I go to Firewall rules ( /cgi-bin/luci/admin/network/firewall/rules ) , and cannot add an SNAT rule, no matter what combination I set there.  I click add and.... Nothing, it just redirects to the rules page.

But if I go to ( /cgi-bin/luci/admin/status/iptables ) I can see that there are some SNAT rules added based on the external WAN port -> LAN port forward rules I added.

reiffert2, 2018-02-18:
I'm on LuCI Master (git-18.039.58622-76f9f5e) / OpenWrt SNAPSHOT r6021-0219138

Clicking on "Add" below "SOURCE NAT" does not change a thing but instead adds the any any rule on the 2nd image.

Do you need more details? Please let me know.

[image]
[image]

wallcreeper, 2018-04-06:
I encounter exactly the same problem as reiffert2 describes. Also I don't have the wan zone.
Is there a way to configure SNAT by console in the mean time the LUCI UI does not provide this funcionality?

I'm on LuCI lede-17.01 branch (git-17.290.79498-d3f0685)

jow-, 2020-02-23:
Master / 19.07.1 gained support for `config nat` sections.

closed 2020-02-23 by jow-