openwrt/luci #925: port forward does not support ipv6

bastien-roucaries, 2017-01-08:
Hi,

The port forward does not support ipv6

cshoredaniel, 2017-01-14:
I believe this is a limitation of the underlying firewall tool OpenWrt/LEDE uses and is not a LuCI issue. @jow-  is that true?

hnyman, 2017-01-22:
Typically you do not need "port forward" with ipv6. As there is typically no NAT, you simply need a traffic rule accepting packets to ipv6 addr XXX / port YYY.

Or are you really talking about IPv6 NAT?

bastien-roucaries, 2017-01-22:
Le 22 janvier 2017 12:02:28 GMT+01:00, Hannu Nyman <notifications@github.com> a écrit :
>Typically you do not need "port forward" with ipv6. As there is
>typically no NAT, you simply need a traffic rule accepting packets to
>ipv6 addr XXX / port YYY.
>
>Or are you really talking about IPv6 NAT?

No i am talking to both: redirection of port 80 to another host and ipv6 nat (nat 1:1) un order to to redirect some port to ula network
-- 
Envoyé de mon appareil Android avec K-9 Mail. Veuillez excuser ma brièveté.

jow-, 2017-01-22:
@cshoredaniel - yes, you're correct. Firewall3 currently does not support `config redirect` for IPv6. This is something I target for LEDE 17.01.1

cshoredaniel, 2017-01-22:
@jow- If you need someone to do the LuCI part (if any is necessary) once you've got the firewall update, I'm happy to help.

LinuxfarmerHH, 2017-10-03:
| luci-app-firewall | git-17.267.26012-a2ea9fd-1

Tried this version today, got an online game with IPv6 support. I can use my IPv6 suffix for the related computer if switched to 'other', but after it is added, luci assign it to ipv4-nat chain.
Ready for beta testing:-)

dermoth, 2018-03-08:
+1

IPv6 has permanent local addresses which can be used to statically address an IPv6 host in a local network. With dynamic DNS I would have to get one domain for each host I want to have external inbound connections to, and have each host update its DNS entry on its own.

Instead I can have the router update its external IPv6 address in dynamic DNS and forward the packets based on dport to the appropriate internal hosts, so externally all services appears to be from the same host.

This also has the advantages of not revealing the real destination host IPv6 addresses.

mrhso, 2018-08-10:
Although IPv6 NAT is not recommend, I use it. Because relay on odhcpd is unstable.
In fact, some ISPs subnetting /128. I think port forward for IPv6 NAT is required.

WRMSRwasTaken, 2018-08-10:
Might be related, as SNAT with v6 does not work either, even after installing `ip6tables-mod-nat`

dmlb2000, 2019-01-06:
Yes, I'm trying to run a transparent squid proxy for HTTP/HTTPS traffic for some hosts on my network. I see the DNAT rules for the host in ipv4 tables but not ipv6 tables. Transparent proxy configuration doesn't even have to have a concept of what IP protocol you are running. I should be able to say, "all traffic from vlan XXX going to wan where destination port is HTTP should DNAT to squid:3129" (https respectively).

n8v8R, 2019-01-28:
Has the issue been abandoned since lingering for 2 years? Or is there some development in master/branch?

moapwr, 2019-01-28:
maybe the overlap between users that care for portforwarding on ipv6 and users that need a helper for it is easily overestimated.

n8v8R, 2019-02-01:
> users that need a helper for it 

That would question of LuCI as helper in general. Sure everything could be done via cli solely but hen why bother with UCI/LuCI in the first place.

dermoth, 2019-02-02:
I have the same thinking as well - even my scripts which control firewall through cli use UCI, and that for multiple reason:

1. Running config is stored in one place, cleaner backups
2. Upgrade is easier - if things change under the hood I don't have to reverse-engineer and update my tools as well
3. Visibility of what is configured exactly from LuCI
4. Less hacks - ex. if you change something in the firewall you also need to add a hook to re-do the change every time the firewall is reloaded; having everything controlled directly by UCI reduce the risk of bugs.

systemcrash, 2024-05-01:
It definitely does now - 9c55500fe8efa309d55f34c21d5ae2bf69fabf06

closed 2024-05-01 by systemcrash