openwrt/luci #926: [openwrt/luci] ifaces.lua - CBI function to write on DynamicList

KarHoey, 2017-01-09:
Based on #918 issue, I wrote the DynamicList .cfgvalue() function, as below:

 -- create seperate field for user input DHCP option 3 (Router) and 6 (DNS)

	local dhcpopt6 = s:taboption("advanced", DynamicList, "dhcp_option", translate("Alternative Default DNS"), translate(""))
	dhcpopt6.datatype = "ip4addr"

	-- ip4addr datatype does not seems work in dhcpopt3
	local dhcpopt3 = s:taboption("advanced", DynamicList, "dhcp_option", translate("Alternative Default Gateway"), translate(""))
	dhcpopt3.datatype = "ip4addr"
	dhcpopt3.default = "0.0.0.0"

	--check dhcp option enum for dhcpopt3 and dhcpopt6 cfgvalue
	function dhcpopt6.cfgvalue(self, section)
		local v = m2.uci:get("dhcp", section, "dhcp_option")
		-- convert the string value to a one-element array
		local a = {}
		local op = 0
		local j = 1
		if v and #v > 0 then

			for i=1,#v,1 do
				-- discard one or more characters after the the comma
				op = v[i]:match("[^,]+")
				if op == "6" then
					-- print one or more characters after the the comma
					a[j] = v[i]:match("[,](.*)")
					j = j + 1

				end
			end

			return a
		end
	end

	function dhcpopt3.cfgvalue(self, section)
		local v = m2.uci:get("dhcp", section, "dhcp_option")
		local a = {}
		local op = 0
		if v and #v > 0 then

			for i=1,#v,1 do
				op = v[i]:match("[^,]+")
				if op == "3" then
					a[1] = v[i]:match("[,](.*)")
				end
			end

			return a
		end
	end
I wrote .cfgvalue() function to read IP from dhcp_option, and sort them out to respectively fields based on their dhcp option enum.

KarHoey, 2017-01-09:
However I'm lost in writing .write() function because either

m2.uci:get("dhcp", section, "dhcp_option") or 0 OR
dhcp_option:formvalue(section) or 0

return only IP address, without the dhcp option enum at the beginning, so I cannot differentiate whether the IP address is belong to dhcpopt3 or dhcpopt6.

I tried writing separate dhcpopt6.write() and dhcpopt3.write() but it don't work as well.

How can I retrieve value from uci files in lua? Thank you.

jow-, 2020-02-23:
That part of LuCI has been entirely rewritten in client side JS.

closed 2020-02-23 by jow-