openwrt/luci #928: luci-ssl-openssl selects libmbedtls and libopenssl

thagabe, 2017-01-10:
Why does luci-ssl-openssl select both these libraries? This seems to just increase file size. Reason why I use openssl is because of the hardware offload it allows.

hnyman, 2017-01-10:
> Why does luci-ssl-openssl select both these libraries? 

The reason is mentioned in the luci-ssl-openssl Makefile:
https://github.com/openwrt/luci/blob/master/collections/luci-ssl-openssl/Makefile

As LuCI is common to both Openwrt and LEDE, luci-ssl-openssl still selects also px5g that brings in mbedtls (in LEDE) or polarssl (in Openwrt).

* uhttpd in LEDE has been patched to generate SSL keys using the openssl cmd (from openssl-util package)
* uhttpd in Openwrt does not yet know how to generate the SSL keys with openssl command line tools. It uses px5g for it.

Luci-ssl-openssl is currently defined as:
` LUCI_DEPENDS:=+luci +libustream-openssl +px5g `

In LEDE it is possible to replace px5g with openssl-util and achieve "openssl-only" build without px5g and libmbedtls, as uhttpd can also generate keys with openssl commandline tools if px5g is not present.

You can achieve openssl-only build two ways:
* edit luci-ssl-openssl Makefile in your LuCI feed repo and replace ` +px5g ` with ` +openssl-util `
* do not select luci-ssl-openssl, but select separately `luci`, `libustream-openssl` and `openssl-util`

(luci-ssl-openssl was introduced to enable using libustream-openssl more easily for luci SSL connections, but luci-ssl-openssl is not perfect as it still includes px5g.)

thagabe, 2017-01-13:
Thank you for the thorough explanation!

closed 2017-01-13 by thagabe

hnyman, 2017-01-31:
Openwrt uhttpd has now been patched, so I committed https://github.com/openwrt/luci/commit/b6341bd197aed9640c4d0dcff73019e045978df0 to remove the dependency for px5g (and mbedtls/polarssl)