openwrt/luci #934: meshwizard: deal/delete new default_radio0 wifi

FreifunkUFO, 2017-01-11:
the wifi config has changed (LEDE only?), so meshwizard has to be changed

using meshwizard needs atm a workaround like this
uci delete wireless.default_radio0
uci commit

FreifunkUFO, 2017-01-11:
i think it was introduced here 
https://github.com/lede-project/source/commit/5e35b4562fb73aaa1c75a3404fed2efb6afe03f8#diff-2bdbd05ca848d5f093d63d6b423a6654

@jow- told at lede-dev mailinglist some idea which has to be done and tested now:

The problem is exactly in the file you
linked:

https://github.com/openwrt/luci/blob/master/contrib/package/meshwizard/files/usr/bin/meshwizard/helpers/setup_wifi.sh#L49

You need to change that "if" statement to something like:

if [ -z "${1/cfg[0-9a-fA-F]*/}" ] || \
[ "$1" = "default_${device}" ]; then

also see https://plan.leipzig.freifunk.net/issues/360

adelton, 2025-01-09:
The meshwizard was moved to https://github.com/freifunk/openwrt-packages in 4654ba92c3999d84b08211f05d2ccb5ea9da731b. This issue should be moved there / opened there if still valid. But since the code no longer appears to exist in this repo, the issue here should likely be closed.