openwrt/luci #941: Build fails: modules/luci-base/Makefile uses "$(1)" instead enviroment variable

grumbler66rus, 2017-01-12:
I try build the OpenWRT today and build failed.
In the modules/luci-base/Makefile (commit 61fa65f2f99ab05302911cab9527411d3132ba61) at the lines 40-42  used the 1st positional parameter but in the host-build.mk (openwrt/include/host-build.mk) the "Host/Install" called without parameters. At result, script try install files to the "/bin" directory.

In this case I simple replace "$(1)" with the real path, because I don't known what variable should be used here. After this, build proceed continued success.

cshoredaniel, 2017-01-14:
@grumbler66rus @hnyman @jow- I've confirmed this isn't the case for LEDE; must be a different build in OpenWrt.

cshoredaniel, 2017-01-14:
@grumbler66rus silly question; what version of OpenWrt (trunk; 15.05?)

hnyman, 2017-01-14:
On 2017-01-14 13:35, Daniel Dickinson wrote:
> @grumbler66rus [1] silly question; what version of OpenWrt (trunk;
> 15.05?)

I suspect 15.05 build trying to use Luci master. And that is 
unsupported.

grumbler66rus, 2017-01-14:
@cshoredaniel, OpenWRT trunk. (And I updated post.)

hnyman, 2017-01-14:
Where are you fetching Openwrt sources?
What is the commit hash of the last commit in Openwrt?

Git.openwrt.org is stale and should not be used
Github.com/openwrt/openwrt has the current sources.

That function call parameter was added in October 2016 and both Openwrt 
trunk and LEDE were fixed to do like OK. Compilation works ok with 
up-to-date sources from github. Or at least worked 2 days ago. But 
git.openwrt.org sources will fail.

See discussion in 
https://github.com/openwrt/luci/commit/4d97a9f6986fb37ce7e69a1ca3f4743334495654

grumbler66rus, 2017-01-15:
@hnyman , you rigth! I used outdated instruction at russian page of OpenWRT Wiki. Today i will fix this wiki page.

closed 2017-01-15 by grumbler66rus