openwrt/luci #944: community-profiles: after sysupgrade old profiles are kept

SvenRoederer, 2017-01-13:
I just wonder why the old profiles are kept, when a system is upgraded?

This results in the fact, that on a freshly upgraded system you may have profiles which contain the data of 4 years ago ...

Is this by design or should we work around this?

mmunz, 2017-01-14:
@SvenRoederer Afair this was no design decision but just forgotten. I think you are right and these files should be excluded from sysupgrade.

hnyman, 2017-01-23:
Well, there is not much you can do, as the profiles are stored in /etc/config , which is always included in the sysupgrade backup. I don't think that you can easily exclude files from there.

(I mostly wonder why they are stored in /etc/config in the first place...
In my opinion it would be more logical to store them e.g. in /etc/freifunk-profiles and then modify scripts to use a suitable uci parameter to point into that directory when those profiles are accessed. Haven't looked into scripts, but that is the first thing that comes into my mind. )

Ps. I briefly looked into the Freifunk stuff and noticed the uci-defaults scripts freifunk-common/files/etc/rc.local.d/01-config-migration . Is that 6-year old settings migration scripts really needed, still in 2017? It e.g refers to lucid that removed from the repo in 2014...

SvenRoederer, 2017-01-23:
regarding the profiles in /etc/config: my idea was to store the files outside (e.g. /etc/freifunk) and to link them to /etc/config.
This will break less or maybe nothing and shows the way to go ...

SvenRoederer, 2019-12-24:
I opened PR https://github.com/freifunk/openwrt-packages/pull/21 in the freifunk-repo for this.

closed 2019-12-24 by SvenRoederer