openwrt/luci #962: Confusing Buttons in /view/admin_uci/changes.htm "Unsaved Changes" Page: consider renaming? Cleanup Page.

NvrBst, 2017-01-20:
This is more opinions, but are all easy to change, and would make the "Unsaved Changes" page a lot clearer to new users. Basic ideas to clean up the /view/admin_uci/changes.htm page.

**REORDER THE BUTTONS:** "Apply & Save", "Apply",  then "Revert"
To match the rest of luci's button ordering which is: "Save & Apply", "Save", "Revert". Except on the changes page, it says "Apply", "Apply & Save" then "Revert". Should order them to the **"Apply & Save", "Apply",  then "Revert"**, to more closely match luci's button theme.

**CONFUSING "Unsaved Changes" BUTTON**: Rename to something else
When you make changes and click the "Save" button it goes into the "UNSAVED Changes" page. This doesn't really make sense that clicking "Save" moves something to "Unsaved". This page is more of a "Unapplied Changes" or "Not Applied Changes" or just a log of changes, which could be just renamed to "Changelog". I would rename it to what it is, just a "Log of Changes", or **"Changelog"** where when u make a change and save it, it goes into the log, and you can "Apply" the changes or "Revert" in that page. This button is named in the theme's header.htm, so someone would have to alter the 4 theme's header.htm name to change this button's name.

**CONFUSING "Apply & Save" BUTTON vs "Apply" BUTTON**: Rename "Apply & Save" to something else
The "Apply & Save" button is very confusing. We already clicked a "Save" button to put changes into this page, and now we have a Log of all the changes, what will "Apply & Save" do over just "Appy". This is a VERY confusing button. "Save" and "Apply" in this sense mean the same thing, since the changes are already "saved".
Upon testing this, what "Apply & Save" does is just Apply the Changes, and CLEARS THE LOG. If you just click "Apply" it'll Apply the changes, but keep the changes in the log. That's the difference between these two buttons. The name of the button should be renamed to match the function.
Assuming you agree that this page is more of a "Log of Changes" or a "Log of Saved Changes" or a "Changelog", then you should renamed the "Apply & Save" to something like **"Apply & Clear Log"**. It would help new Luci people understand what is going on.

cshoredaniel, 2017-01-20:
@NvrBst I generally agree with your points, except that Apply & Save is really (Apply & Save OR just Apply if already saved; i.e. you don't have to hit Save before Apply & Save).  In that case the name for the button is about as good as it gets while being sufficiently short & sweet.

cshoredaniel, 2017-01-20:
@NvrBst @jow- @hnyman My biggest concern with the second suggested changes is the lack of translation activity for most translations.  What do you think?

NvrBst, 2017-01-20:
Ah, I'm only talking about the buttons in the "Unsaved Changes" Page. The "Save & Apply" vs "Save" buttons (Everyone else) I understand.

However in the "Unsaved Changes" Page, top right (after you have just saved stuff, but not applied stuff).  Has a different button named "Apply & Save" This is the confusing one, since everything on this page is "Saved already". In fact this button really doesn't Save, it "Apply & Clears".

cshoredaniel, 2017-01-20:
@NvrBst Oops, sorry, misundersood.  I agree completely then.  Just changing it to Apply I think would be best (I think saying clear might actually suggest some kind of reset, so best just say apply).

NvrBst, 2017-01-20:
Ah yes,  But there is already an "Apply" button, and also an "Apply & Save". So in the "Unsaved Pages" You have a log of all the "Saved" stuff you did, and you have a choice of 3 buttons to click. I do not think most people know what the difference between the first 2...
1. **Apply & Save**: Most would assume this applies all the changes. But what does the save part mean? All these changes are SAVED already...
2. **Apply**: Most would assume this does the same as 1... 
3. **Revert**: This everyone would understand.

With testing, to me, it seems "Apply" Applies all the changes, but leaves all the changes in this log so u can see them and revert them if u wanted (the "Unsaved Changes" button remains). "Apply & Save" Applies all the changes, and then removes the log. I'm unsure what the "Apply & Save" should be named to, but pretty sure "Apply & Save" isn't correct. It should be something like "Apply & Close" or "Apply & Clear Log". Basically it needs to describe the function of:

- **"Apply & Save"**: Applying all the Saved Changes, and CLEARS the current Log of Changes you have.
- **"Apply"**: Applying all the Saved Changes, but keeps the current "Log of Changes" you have, so the "Unsaved Changes" button is still in the top right of the screen even after you hit "Apply". Of course hitting Apply again on this page will "reapply" the current list if nothing new was changed. This button is not so useful, but maybe for some people.

[Picture of the "Unsaved Pages" page](https://s29.postimg.org/bxs6usc6v/image.png)

jow-, 2017-01-20:
It would proably make sense to rename it to "Apply & Commit". In uci terms, "save" means to stage pending changes into a temporary directory which is gone after a reboot (or revert) while "commit" writes staged changes into the persistent storage.

cshoredaniel, 2017-01-22:
@hnyman would you regard this as a translation breakage issue if it were done but only for trunk?

hnyman, 2017-01-22:
> would you regard this as a translation breakage issue if it were done but only for trunk?

Hmm. Not quite sure what you mean. 
But principally the same answer as in the other topic: if we change existing key button labels on core functionality, it would be great to edit .po files with sed (or manually) to ensure that most users do not see other buttons in local language and one in English.

In this case I think that there would still be a "Save & Apply" string (that has been there already earlier) and then a new "Apply & Commit" replacing S&A on the unsaved changes page, which string is new and untranslated. So no breakage.

cshoredaniel, 2017-01-22:
@hnyman I actually meant 'do we need to do the edit .po files, etc thing'.  I call it breakage because AIUI the string becomes untranslated and therefore not in the target language, which is translation breakage vs. software breakage.

cshoredaniel, 2017-01-22:
Not sure when I'd get to this, so unassigning; thinking just add a 'confirmed' label just to be able indicate it's triaged ( @hnyman would you think that okay )?

NvrBst, 2017-01-30:
The current "Apply & Save" string on the "Unsaved Changes" page is not in the .po files, and thus not translated. So changing that to "Apply & Commit" will be fine, and no .po file changes will be necessary. (there is the "Save & Apply" (standard), but no "Apply & Save" (unsaved changes page) translation).

I like "Apply & Commit". It's much more obvious what is going on than what is there currently. The only issue is that the "Apply" on this page doesn't match the "Apply" everywhere else in LuCI (the Apply everywhere else actually means Commit). So if the common user has been using the standard "Save & Apply" button, he may be confused what the difference between "Apply" and "Apply & Commit" on this page. Since the "Save & Apply" he's been using all this time is actually a "Save & Commit". I assume a "Save & Apply" === "Commit". So really this page is just a "Commit" / "Apply" / "Revert". But introducing a solo "Commit" definition may confuse people, where an "Apply & Commit" shows that this button is doing one more action over the solo "Apply" button.

Changing the "Unsaved Changes" button would require modifying 16 *.po files (.po files with non-blank translations). I speak ca/french/jap and my brother german/spanish. So I can do 5/16 *.po properly. The other 11 I would have to rely on Google Translate. Looking quickly at the translations, it seems some of them are translating the buttons meaning over the literal text anyway. For example, the French Canadian button is translated "Unsaved Changes", while the France French version is "Not Applied Changes", which is actually what the button is. I assume some other nationalities have done the same and translated the buttons function and not the "text".
Would still need to choose a name for this though over "Unsaved Changes". Some suggestions would be "Saved Changes", "Unapplied Changes", "Modified Changes", "Uncommitted Changes" or simply just "Changes".   "Log of Changes" or "Changelog" also are options... Unapplied or Uncommitted or just Changes, seem like the more logical options maybe. After clicking "Save", and seeing "Saved Changes" at the top may confuse some users that the changes are actually Applied. After clicking "Save" and seeing "Unsaved Changes" at the top is probably more confusing to some though.

feckert, 2022-12-16:
Works as expected and would not to get changed in the future

closed 2022-12-16 by feckert