openwrt/luci #963: BUG: DHCP Leases "infinite" time shows as "expired" not as "unlimited" in IPv4

NvrBst, 2017-01-20:
I'm on "LEDE Reboot SNAPSHOT r3025-c99f881 / LuCI Master (git-17.017.13638-9b7a5a4)", Router: Archer C7. And in this version of the latest LEDE, the DHCP IPv4 Leases which are set to "infinite" are showing as "expired". The IPv6 leases are fine, and showing as "unlimited", but the IPv4 ones show as "expired".

I see jow tried to fix this already, and the code in these two files below (also most of this is the same code, the admin_status should just call the lease_status, not copy the code into it).
_luci/modules/luci-mod-admin-full/luasrc/view/admin_status/index.htm
luci/modules/luci-mod-admin-full/luasrc/view/admin_network/lease_status.htm_
```
for( var i = 0; i < info.leases.length; i++ )
{
	var timestr;
	if (info.leases[i].expires === false)
		timestr = '<em><%:unlimited%></em>';
	else if (info.leases[i].expires <= 0)
		timestr = '<em><%:expired%></em>';
	else
		timestr = String.format('%t', info.leases[i].expires);
```

The commit was done here: [https://github.com/openwrt/luci/commit/70bad4da4fcd44715445c36d8c4b9f8a00d92b4d](https://github.com/openwrt/luci/commit/70bad4da4fcd44715445c36d8c4b9f8a00d92b4d)

But at least on the latest LEDE, the infinite IPv4 leases.expires property is not getting set to false, and are showing up as expired. The "info.leases[i].expires" property has a value _<-1470000000_, and keeps going down as time runs.

Like I said the IPv6 does show as unlimited on my router, but the IPv4 infinite leases are not.

cshoredaniel, 2017-01-20:
That is actually a consequence of using dnsmasq for DHCPv4; dnsmasq doesn't report infinite leases as -1, but as expiring the maximum time in the future it can set.  If you use odhcpd for DHCPv4 this will work as it should.  Not a LuCI bug.

NvrBst, 2017-01-20:
Ah, thanks @cshoredaniel ! Good to know.

While in the latest LEDE snapshot r3025, it seems to use dnsmasq by default for IPv4? Or one of the apps set it? I just flashed the latest LEDE, then ran the commands below. I didn't explicitly change to dnsmasq.
```
opkg update
opkg install luci ipset tor ip-tiny
opkg install luci-theme-material luci-app-upnp luci-app-ddns luci-app-qos
```

If _dnsmasq_ is the default for OpenWRT/LEDE, perhaps the unlimited lease check in LuCI should check the dnsmasq case? Perhaps a check if it's an insanely large negative number? or is there any other way?
`if (info.leases[i].expires === false || info.leases[i].expires < -1470000000)`

cshoredaniel, 2017-01-20:
odhcpd is the default for ipv6 (I'm not sure if dnsmasq for ipv6 PR has been accepted yet) and dnsmasq is the default for ipv4.  To use odhcpd for ipv4 set 'option maindhcp 1' in the 'config odhcpd' section of '/etc/config/dhcp', or do 'uci set dnsmasq.@odhcpd[-1].maindhcp=1; uci commit dhcp'

cshoredaniel, 2017-01-20:
@NvrBst I'm not sure exactly what dnsmasq reports; it may actually report the time as now - 1 or some such silliness.  I'd have to check /tmp/dhcp.leases and do date -d '@<epoch-from-tmp-dhcp.leases>'. the epoch is the large unsigned integer in /tmp/dhcp.leases and represents the number of seconds since the start of the unix epoch (a historical legacy date when it was first defined).

NvrBst, 2017-01-20:
I see! Upon looking, _dnsmasq_ stores infinite leases as "0" in the _/tmp/dhcp.leases_ file. This corresponds to the _01/01/1970 00:00:00 GMT_, in Unix EPOCH time. Since today's date is "1484889355". But the expires property is (EXPIRY DATE - TODAYS DATE). So it's 0-1484889355. So any value below -1484889355 works.

So if someone wanted to do a dnsmasq infinite check properly you could:

1. Check the _/tmp/dhcp.leases_ file and see if it has a 0 as the time.
2. Since the DHCP Lease Table is pulling its information from:

3. `XHR.poll(5, '<%=url('admin/network/dhcplease_status')%>'`
4. Going to the URL: _http://192.168.0.1/cgi-bin/luci/admin/network/dhcplease_status_ shows that it only gives the expiry conversion, and not the original expiry date. So someone would have to alter the structs and put the expiry time, and original EXPIRY DATE. Then you can use that and check to see if it's 0 for an infinite lease --- More work.
5. The leases[i].expires property is just EXPIRY DATE - CURRENT DATE. So you could reverse this calculation and check to see if it's 0. Of course, playing with time can be smudgy with rounding.

Option 2 is probably the cleanest, Option 3 is the easiest. So if we were doing 3, we can just do the check in these two files as:

_luci/modules/luci-mod-admin-full/luasrc/view/admin_status/index.htm
luci/modules/luci-mod-admin-full/luasrc/view/admin_network/lease_status.htm_
```
					if (info.leases[i].expires === false || Math.trunc((new Date().getTime()/1000 + info.leases[i].expires)/10) == 0)
						timestr = '<em><%:unlimited%></em>';
					else if (info.leases[i].expires <= 0)
						timestr = '<em><%:expired%></em>';
```

cshoredaniel, 2017-01-20:
> Mention
> <mention@noreply.github.com> Subject: Re: [openwrt/luci] BUG: DHCP
> Leases "infinite" time shows as "expired" not as "unlimited" in IPv4
> (#963)

Sent the following message to odhcpd maintainer and lede-dev mailing
list.

Hi Hans,

In LuCI GitHub issue 963 (https://github.com/openwrt/luci/issues/963),
a user discovered that LuCI incorrectly reports dnsmasq infinite
leasetime as expired because the LuCI check is based on odhcpd's -1 for
INFINITE_VALID, but dnsmasq uses 0 in the dhcp.leases file for
infinite leases.  Would it be preferable to sync odhcpd and dnsmasq
(I know odhcpd at one time used 0 for INFINITE_VALID in the statefile
but that that was changed, and I don't know why).  From a LuCI point of
view I'm not sure if there some caveat with the 0 value (I seem to
recall odhcpd uses 0 to represent something other than infinite, but I
could be wrong).  Before hacking on LuCI for the dnsmasq case I want to
make sure I understand what values are going to be found in the odhcpd
case.

User's findings (last comment only) reproduced below.

Regards,

Daniel

Begin forwarded message:

Date: Thu, 19 Jan 2017 22:38:34 -0800

https://github.com/openwrt/luci/issues/963#issuecomment-273989416

cshoredaniel, 2017-01-20:
As you can tell it remains to be seen whether the fix is best done in core or here.

NvrBst, 2017-01-20:
I see! Upon looking, _dnsmasq_ stores infinite leases as "0" in the _/tmp/dhcp.leases_ file. This corresponds to the _01/01/1970 00:00:00 GMT_, in Unix EPOCH time. Today's date is "1484889355",  but the _leases[i].expires_ property is (EXPIRY DATE - TODAYS DATE). So it's 0-1484889355. Thus any value below -1484889355'ish is infinite.

So if someone wanted to do a dnsmasq infinite check properly you have several options:

1. Check the _/tmp/dhcp.leases_ file and see if it has a 0 as the time.
2. Since the DHCP Lease Table is pulling its information from:
`XHR.poll(5, '<%=url('admin/network/dhcplease_status')%>'`
Going to the URL: _http://192.168.0.1/cgi-bin/luci/admin/network/dhcplease_status_ shows that it only gives the expiry conversion, and not the original expiry date. So someone would have to alter the structs and put the expiry time, and original EXPIRY DATE into _dhcplease_status_ . Then you can use that and check to see if it's 0 for an infinite lease --- more work.
3. The _leases[i].expires_ property is just EXPIRY DATE - CURRENT DATE. So you could reverse this calculation and check to see if it's 0. Of course, playing with time can be smudgy with rounding.

Option 2 is probably the cleanest, Option 3 is the easiest. So if we were doing 3, we can just do the check in these two files as:

_luci/modules/luci-mod-admin-full/luasrc/view/admin_status/index.htm_
```
var ls = document.getElementById('lease_status_table');
if (ls)
{
	var localEpoch = new Date(info.localtime).getTime()/1000.0;
...
	for( var i = 0; i < info.leases.length; i++ )
	{
		var timestr;

		if (info.leases[i].expires === false || Math.trunc(localEpoch + info.leases[i].expires) == 0)
			timestr = '<em><%:unlimited%></em>';
		else if (info.leases[i].expires <= 0)
			timestr = '<em><%:expired%></em>';
```

_luci/modules/luci-mod-admin-full/luasrc/view/admin_network/lease_status.htm_
```
var tb = document.getElementById('lease_status_table');
if (st && st[0] && tb)
{
	var curEpoch = new Date().getTime()/1000.0;
...
	for( var i = 0; i < st[0].length; i++ )
	{
		var timestr;

		if (st[0][i].expires === false || Math.trunc((curEpoch + st[0][i].expires)/10) == 0)
			timestr = '<em><%:unlimited%></em>';
		else if (st[0][i].expires <= 0)
			timestr = '<em><%:expired%></em>';
```

For admin_status, we have _"info.localtime"_ as a seed for the Date(info.localtime). When using this, it seems like adding info.leases[i].expires to it (and truncating) produces 0 all the time. It's very accurate.

For lease_status.htm, we do not have localtime variable to see, so we needed to use the currentTime. With testing, _var curEpoch = new Date().getTime()/1000.0;_ was always about 1 to 2 seconds off of _info.leases[i].expires_. So I added a +/-10s check to that one. Maybe it's best to add the +/-10s check to both, I'm not too sure.

The first part of the check is the _odhcpd_ version of the check. The second part checks the _dnsmasq_ version. This just sees if the _info.leases[i].expires_ is within +/-10s of _01/01/1970 00:00:00 GMT_, which is 0 in EPOCH time. I did the +/- 10s, because the Date() function was not called at the same time as the leases.expires was calculated. In the admin_status page we had a seed and the +/-10s check may not be nessasary?

This works. I only did this check for the IPv4 part, since _dnsmasq_ is only used by default with the IPv4. I left the IPv6 check as it was.

Of course if there is an easy way to get the Original Expiry Date from _/tmp/dhcp.leases_, or put the original date in the _'admin/network/dhcplease_status'_, then it would be best to use that and check if it's 0. Else the above works, and all you need to do is call one extra Date() call per table.

NvrBst, 2017-01-20:
ah thanks daniel! I deleted my last comment and made a new one with just taking the "new Date()" call out of the for loop, cuz it didn't need to be in there.

But yea best to see if they can sync the _odhcpd_ and _dnsmasq_ cases. Thanks again for all your patience.

cshoredaniel, 2017-01-20:
@NvrBst and I appreciate your taking the time to look further; there is only so much the regular devs can do in the time/resources we are able to contribute.

scineram, 2017-01-21:
Shouldn't by default odhpcd be used for all DHCP? And leave the DNS to qnsmasq? Is there some special reason to keep using dnsmasq for DHCPv4?

cshoredaniel, 2017-01-21:
@scineram That's the decision made by upstream; discussion of that belongs on http://lists.infradead.org/mailman/listinfo/lede-dev

cshoredaniel, 2017-01-21:
and you also have to coordinate with https://lists.openwrt.org/cgi-bin/mailman/listinfo/openwrt-devel

cshoredaniel, 2017-01-21:
Begin forwarded message:
@NvrBst Sadly this mya not be resolvable.

Date: Fri, 20 Jan 2017 08:59:23 +0100
From: Hans Dedecker <dedeckeh@gmail.com>
To: Daniel Dickinson <lede-daniel@cshore.thecshore.com>
Cc: LEDE Development List <lede-dev@lists.infradead.org>
Subject: Re: [LEDE-DEV] Fw: [openwrt/luci] BUG: DHCP Leases "infinite"
time shows as "expired" not as "unlimited" in IPv4 (#963)

Hi Daniel

odhcpd uses 0 to indicate a lease is expired while -1 is used to
reperesent infinite leasetime since the support of infinite leasetime
(http://lists.infradead.org/pipermail/lede-dev/2016-November/004133.html).
I have my doubts can you make a distinction in dnsmasq between
infinite leasetime and a lease which is expired knowing the code
quality of dnsmasq.
So I would prefer to keep -1 for infinite leasetime to make it clear
it's "infinite" and to make a distinction with an expired lease.

Hans

_______________________________________________
Lede-dev mailing list
Lede-dev@lists.infradead.org
http://lists.infradead.org/mailman/listinfo/lede-dev

cshoredaniel, 2017-01-21:
Closing unless someone decides to work on dnsmasq and make this work.

NvrBst, 2017-01-22:
Sometime in the future _dhcpv4_ will prob switch _odhcp_, and everything will work fine anyway.

Although, I don't think the way to fix this is to make _dnsmasq_ mimic _odhcp_, but rather just have LuCI check _odhcp_ and _dnsmasq's_ conditions for an infinite lease. Which is just seeing if the lease expiry date is set to _01/01/1970 00:00:00 GMT_ (i.e. current time + the expiry time == 0).

```
var localEpoch = new Date(info.localtime).getTime()/1000.0;
...
if (info.leases[i].expires === false || Math.trunc(localEpoch + info.leases[i].expires) == 0)
```

It's only adding 2 lines of code to have LuCI check for _dnsmasq_ infinite leases for the _dhcpv4_ case. If you use currentEpoch, you need to check within +/-10s.

cshoredaniel, 2017-01-22:
@NvrBst Unfortunately from what Hans said, dnsmasq using 0 for expired strings as well as infinite strings; i.e. there is no differentiation based on his statement.

NvrBst, 2017-01-22:
> I have my doubts can you make a distinction in dnsmasq between infinite leasetime and a lease which is expired knowing the code quality of dnsmasq.

I think he is just saying that both cases are just expired leases in dnsmasq. So it may be difficult.
In fact, an **infinite lease in dnsmasq is just a "lease with an expiry date of 01/01/1970 00:00:00 GMT"**. And an **expired lease is just a "lease with an expiry date <= current date (and =! to above date)"**.

If the above is true, one can most definitely distinguish the different between infinite vs expired in dnsmasq by checking the expiry date.

So to test an expired lease you just test to see if the expiry date is <= current date. And if it's expired you see if the expiry date is = 01/01/1970 00:00:00 GMT, if so, it's not expired it's "infinite". If not, then it's just expired. _NOTE: this test could also be done by looking at the /tmp/dhcp.leases file, and seeing if the EXPIRY DATE (first column = 0) which means EXPIRY DATE=01/01/1970 00:00:00 GMT. Problem is LuCI doesn't expose the EXPIRY DATE to us, just the REMAINING EXPIRY TIME. So you have to re-calculate the date._

That said, the 2 code changes above is distinguishing between infinite and expired leases in DHCPv4, as you can see in the below screenshots.

The top 2 leases are "infinite leases" and you can see this by the leases file = 0, below that. I set my lease time to 2m, and requested a lease on LAPTOP-NGG, then disconnected wifi. After the countdown occurred, the lease went to "expired". And the top 2 remained "unlimited". You can see in the dhcp.lease file, the unlimited leases has a DATE (first column) as 0, which is 01/01/1970 00:00:00 GMT, the expired lease has a date of 1485079535, and it remained that. which was a date that became lower than current EPOCH date. Note that shortly after this dnsmasq removed the expired lease from the leases file and thus LuCI remove the entry as well.
[image]
[image]

NvrBst, 2017-01-22:
https://github.com/openwrt/luci/blob/70bad4da4fcd44715445c36d8c4b9f8a00d92b4d/modules/luci-base/luasrc/tools/status.lua#L32

Else you can change this file to alter the ".expires" property of the dnsmasq part. I can see jow changed this file to handle the odhcpd infinite case last month (infinite = -1, return false). You can change line 32 to just add a check if "ts" == 0, then return false (i.e. if expiry date = 1970 date then return "false=infinte") -- if it's not 0, then do what you do now, calcualte difference, if negative then expired, if not, then valid. The code change is below, mimics the odhcp case, just with a differnt check.
```
local expire = tonumber(ts) or 0
expires  = (expire ~= 0) and os.difftime(expire, os.time()),
```

I tested this on my system as well, and again it distinguishes between infinite and expire. And you do not have to change any of the checks in the HTML part. What's nice about this is that it makes the checks the same in the odhcp and dnsmasq case (.expires property is == false if infinite,  <= 0 if expired).

Screenshot of my system with the above change, and unaltered HTML status/dhcplease page.
[image]

hnyman, 2017-01-22:
> The code change is below, mimics the odhcp case, just with a different check.
> 
```
local expire = tonumber(ts) or 0
expires  = (expire ~= 0) and os.difftime(expire, os.time()),
```

That looks nice and elegant.

cshoredaniel, 2017-01-22:
@NvrBst @hnyman Thanks for clarify; I clearly misread Hans comments.

cshoredaniel, 2017-01-22:
I probably won't get to this for a while so I've unassignmed myself and marked it 'help wanted' in hopes someone else is willing and able to spin a patch.

hnyman, 2017-01-23:
Fixed with https://github.com/openwrt/luci/commit/0f35de886c98404f1bf75460f7a76b6cb463e1e1

closed 2017-01-23 by hnyman