openwrt/luci #968: Issue & PR list is hard to read do to excessive amount of colorful classification tags

hnyman, 2017-01-22:
@cshoredaniel apparently likes Github classification tags, and currently there are so many colorful tags added to each issue and PR, that the issue and PR list pages are very hard to read :-(

At least for me the colorful tags draw the attention away from the issue / PR subjects, which get hidden in the colorful classification tag clutter. There are 4-5 tags per issue, in several case overflowing to the next line in the list.

I tried changing some of the tag colors to more muted pastel colors before my vacation, but during my vacation some new colorful classification have been created and applied :-( 

I propose that we 
* limit a bit the use of classification tags. 1-2 most important tags per issue, like "work needed" etc. mainly from status class "B:". And the colors should be more muted.
* using the module "C:" tags for pointing the luci respective module class is usually unnecessary as it is self-evident.
* Priority "P:" tags may be used in some cases, but also they are largely unnecessary
* Some tags are overlapping, e.g. "B: needs testing", "B: ready for further review"

cshoredaniel, 2017-01-22:
@hnyman Points taken; I have a rather large window so I don't notice the size issue.  I didn't realize what  you were doing with the pastels.  I agree the colours (which were already in the label definitions) are overbright and should be muted.  As you may have notice I like classifying things, but as it creates problems as applied, I agree we should stick to 1-2 take (as you say the B: tag (and drop the B:)).  I'd also keep the 'not a luci bug', 'not a bug' (or use invalid for that), wontfix.  The C tags were more because they were there but are definitely not required, and really I think only 'critical' category is needed. 'needs testing' was made separate from 'ready for further review', to differentiate between not tested yet, and tested, but wanting additional opinions. (The review was in the sense of getting more opinions vs. testing, without so much clutter in the comments).

So to summarize, if you agree, I will:

1) Drop C: tags
2) Drop all P: tags except critical (renamed to just critical)
3) Drop B: from B: tags.
4) Drop T: tags

I'll let you choose the colours though; I didn't choose the old ones, but I didn't find them all that disturbing either, so you're a better judge of what suits your needs.

cshoredaniel, 2017-01-22:
@hnyman Could you look at the new label list and see if this look better to you (I haven't done 1,2,4,5 and did do some colour changes, wanted to know what you think)?

cshoredaniel, 2017-01-22:
@hnyman one modifcation; will keep the C: tags but only use them if it's not clear and helps to have the tag.

hnyman, 2017-01-22:
Looks mainly good to me.

cshoredaniel, 2017-01-22:
I think for some of the tags, all I'm really looking for is some indication of whether it's a never looked at bug/PR or whether it awaits other devs input, or whether, triager can't work on it, so either another dev needs to claim it, or a community member help (I think 'help wanted', works for that).  The goal is to be able to quickly get some important information without really reading the title or opening the PR/Issue (and to do searches/filters).

cshoredaniel, 2017-01-22:
@hnyman Could you review the modifcations I've made and see if you are happier?

hnyman, 2017-01-22:
Much better. Big thanks.

closed 2017-01-22 by hnyman