openwrt/luci #985: luci-app-statistics sensors - missing graphs

kofec, 2017-02-01:
@jow- 
You did a great job adding sensors for luci https://github.com/openwrt/luci/commit/cff2b99b4f9af0957ff6de4fe1ce48d30337b41b but one small thing is missing. You scan for all sensors and gather all data. But there is no definition for graphs - only for temperature. Maybe you can definitions for Volatge and fan speed graphs.
applications/luci-app-statistics/luasrc/statistics/rrdtool/definitions/sensors.lua
Thanks in advance

jow-, 2017-02-01:
@kofec - I know, in the commit message I wrote:

> Please send the output of "rrdtool info /tmp/rrd/*/sensors-*/*.rrd"
> if your system happens to support voltage, power or fanspeed sensors.

Do you have a system that supports voltage and/or fan speed? If so can you provide the output?

kofec, 2017-02-02:
Yes, i wilk do this today

kofec, 2017-02-02:
@jow- 
I attach file - https://github.com/kofec/openwrt-n2n/blob/master/files/sensors-lm85-i2c-0-2e.zip
simple output:
```
root@LEDEnsa310:~# sensors
lm85-i2c-0-2e
Adapter: mv64xxx_i2c adapter
in0:          +1.80 V  (min =  +0.00 V, max =  +3.32 V)
in1:          +1.08 V  (min =  +0.00 V, max =  +2.99 V)
in2:          +3.21 V  (min =  +0.00 V, max =  +4.38 V)
in3:          +5.05 V  (min =  +0.00 V, max =  +6.64 V)
in4:         +12.19 V  (min =  +0.00 V, max = +15.94 V)
fan1:        1009 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
fan4:           0 RPM  (min =    0 RPM)
temp1:        +27.0°C  (low  = -127.0°C, high = +127.0°C)
temp2:        +30.0°C  (low  = -127.0°C, high = +127.0°C)
temp3:        +34.0°C  (low  = -127.0°C, high = +127.0°C)
cpu0_vid:    +0.000 V
```
At the beginning I monitor all sensors but after saome time I change it and monitor only in4, fan1 and all temp.
output of your command:

```
root@LEDEnsa310:~# /usr/sbin/sensors -u -A
lm85-i2c-0-2e
in0:
  in0_input: 1.797
  in0_min: 0.000
  in0_max: 3.320
  in0_alarm: 0.000
in1:
  in1_input: 1.078
  in1_min: 0.000
  in1_max: 2.988
  in1_alarm: 0.000
in2:
  in2_input: 3.214
  in2_min: 0.000
  in2_max: 4.383
  in2_alarm: 0.000
in3:
  in3_input: 5.052
  in3_min: 0.000
  in3_max: 6.641
  in3_alarm: 0.000
in4:
  in4_input: 12.188
  in4_min: 0.000
  in4_max: 15.938
  in4_alarm: 0.000
fan1:
  fan1_input: 3112.000
  fan1_min: 0.000
  fan1_alarm: 0.000
fan2:
  fan2_input: 0.000
  fan2_min: 0.000
  fan2_alarm: 0.000
fan3:
  fan3_input: 0.000
  fan3_min: 0.000
  fan3_alarm: 0.000
fan4:
  fan4_input: 0.000
  fan4_min: 0.000
  fan4_alarm: 0.000
temp1:
  temp1_input: 27.000
  temp1_max: 127.000
  temp1_min: -127.000
  temp1_alarm: 0.000
  temp1_fault: 0.000
temp2:
  temp2_input: 30.000
  temp2_max: 127.000
  temp2_min: -127.000
  temp2_alarm: 0.000
temp3:
  temp3_input: 34.000
  temp3_max: 127.000
  temp3_min: -127.000
  temp3_alarm: 0.000
  temp3_fault: 0.000
cpu0_vid:
  cpu0_vid: 0.000

root@LEDEnsa310:~#
```

kofec, 2017-03-04:
@jow- 
Are the logs OK ? Any more is needed ? Maybe I should provided in other format ?

BlueMax, 2019-06-02:
@jow- Here is the output you requested. Hope you can fix it.
`$ for i in /tmp/rrd/OpenWrt/sensors-f71808a-isa-0290/*.rrd; do rrdtool info $i >> /tmp/rrdtool.txt; done`

[rrdtool.txt.tar.gz](https://github.com/openwrt/luci/files/3244350/rrdtool.txt.tar.gz)

feckert, 2019-10-07:
@kofec is this still an issue?
There were many changes in the master of this app

kofec, 2019-10-07:
I will check this today

kofec, 2019-10-09:
Last time @jow- mentioned that it is not implemented. I try it now and nothing changed. I think It is not a bug rather the new feature :-) :
```
root@NSA310:~# ls -l /tmp/rrd/NSA310/sensors-lm85-i2c-0-2e/
fanspeed-fan1.rrd
fanspeed-fan2.rrd
fanspeed-fan3.rrd
fanspeed-fan4.rrd
temperature-temp1.rrd
temperature-temp2.rrd
temperature-temp3.rrd
voltage-in0.rrd
voltage-in1.rrd
voltage-in2.rrd
voltage-in3.rrd
voltage-in4.rrd

```

ThBexx, 2021-11-11:
I can confirm above comments / output / results on OpenWrt 21.02.1
Hardware (like above): Zyxel NSA310

Thank you for the great work / implementation. Can you add the functionality to display RPM and voltages?

Output of `rrdtool info /tmp/rrd/nas310/sensors-lm85-i2c-0-2e/*` + `*.rrp files`
[NSA310_lm85_sensor.zip](https://github.com/openwrt/luci/files/7523501/NSA310_lm85_sensor.zip)

### RRDtool 1.0.50
```
root@nas310:/tmp/rrd/nas310/sensors-lm85-i2c-0-2e# ls -lsh
     8 -rw-r--r--    1 root     root        6.9K Nov 11 22:57 fanspeed-fan1.rrd
     8 -rw-r--r--    1 root     root        6.9K Nov 11 22:23 fanspeed-fan2.rrd
     8 -rw-r--r--    1 root     root        6.9K Nov 11 22:23 fanspeed-fan3.rrd
     8 -rw-r--r--    1 root     root        6.9K Nov 11 22:23 fanspeed-fan4.rrd
     8 -rw-r--r--    1 root     root        6.9K Nov 11 22:57 temperature-temp1.rrd
     8 -rw-r--r--    1 root     root        6.9K Nov 11 22:57 temperature-temp2.rrd
     8 -rw-r--r--    1 root     root        6.9K Nov 11 22:57 temperature-temp3.rrd
     8 -rw-r--r--    1 root     root        6.9K Nov 11 22:57 voltage-in0.rrd
     8 -rw-r--r--    1 root     root        6.9K Nov 11 22:57 voltage-in1.rrd
     8 -rw-r--r--    1 root     root        6.9K Nov 11 22:57 voltage-in2.rrd
     8 -rw-r--r--    1 root     root        6.9K Nov 11 22:57 voltage-in3.rrd
     8 -rw-r--r--    1 root     root        6.9K Nov 11 22:57 voltage-in4.rrd

```

### sensors version 3.6.0 with libsensors version 3.6.0
```
root@nas310:/tmp/rrd/nas310/sensors-lm85-i2c-0-2e# sensors
lm85-i2c-0-2e
Adapter: mv64xxx_i2c adapter
in0:           1.80 V  (min =  +0.00 V, max =  +3.32 V)
in1:           1.09 V  (min =  +0.00 V, max =  +2.99 V)
in2:           3.21 V  (min =  +0.00 V, max =  +4.38 V)
in3:           5.08 V  (min =  +0.00 V, max =  +6.64 V)
in4:          12.31 V  (min =  +0.00 V, max = +15.94 V)
fan1:        3069 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
fan4:           0 RPM  (min =    0 RPM)
temp1:        +30.0°C  (low  = -127.0°C, high = +127.0°C)
temp2:        +32.0°C  (low  = -127.0°C, high = +127.0°C)
temp3:        +35.0°C  (low  = -127.0°C, high = +127.0°C)
cpu0_vid:    +0.000 V
```
```
root@nas310:# /usr/sbin/sensors -u -A
lm85-i2c-0-2e
in0:
  in0_input: 1.797
  in0_min: 0.000
  in0_max: 3.320
  in0_alarm: 0.000
in1:
  in1_input: 1.090
  in1_min: 0.000
  in1_max: 2.988
  in1_alarm: 0.000
in2:
  in2_input:
[NSA310_lm85_sensor.zip](https://github.com/openwrt/luci/files/7523498/NSA310_lm85_sensor.zip)
 3.214
  in2_min: 0.000
  in2_max: 4.383
  in2_alarm: 0.000
in3:
  in3_input: 5.078
  in3_min: 0.000
  in3_max: 6.641
  in3_alarm: 0.000
in4:
  in4_input: 12.250
  in4_min: 0.000
  in4_max: 15.938
  in4_alarm: 0.000
fan1:
  fan1_input: 3061.000
  fan1_min: 0.000
  fan1_alarm: 0.000
fan2:
  fan2_input: 0.000
  fan2_min: 0.000
  fan2_alarm: 0.000
fan3:
  fan3_input: 0.000
  fan3_min: 0.000
  fan3_alarm: 0.000
fan4:
  fan4_input: 0.000
  fan4_min: 0.000
  fan4_alarm: 0.000
temp1:
  temp1_input: 30.000
  temp1_max: 127.000
  temp1_min: -127.000
  temp1_alarm: 0.000
  temp1_fault: 0.000
temp2:
  temp2_input: 31.000
  temp2_max: 127.000
  temp2_min: -127.000
  temp2_alarm: 0.000
temp3:
  temp3_input: 35.000
  temp3_max: 127.000
  temp3_min: -127.000
  temp3_alarm: 0.000
  temp3_fault: 0.000
cpu0_vid:
  cpu0_vid: 0.000
```
[image]

adelton, 2025-01-10:
Does `Monitor all except specified` in latest versions achieve what you are after?