openwrt/luci #996: luci: realtime graphs: realtime connections: prints 'localhost' when dnsmasq not installed

EricLuehrsen, 2017-02-10:
This is a little odd and I haven't had time to figure out exactly why. Its probably obvious but I am just missing it. When dnsmasq **is installed** with or without Unbound (or Bind), then the Realtime Connections graph and list do **not** include localhost (::1,127.0.0.0/8) connections. When **only Unbound** (or Bind) are installed and dnsmasq is not installed, then localhost connections are listed and graphed. These are not meaningful because a lot of them a generated by the graph functions. It would also be good to have consistency regardless of the DNS server.

hnyman, 2017-02-10:
It doesn't look to be that obvious. I looked into this a few minutes, and have some educated guesses and additional questions for you... 

* If it is related to just having dnsmasq installed, it might be due to dependencies of dnsmasq and there might even be a difference between dnsmasq variants. Are you using dnsmasq or dnsmasq-full?

  Dnsmasq-full has a dependency for libnetfilter-conntrack, so if you use -full, that is at least one difference. (But even with my own normal dnsmasq, I do not see localhost-localhost connections listed)

* The second alternative is that dnsmasq init script does something to the conntrack settings even when dnsmasq is configured to do nothing.

* Third alternative is again just a hunch, but the luci realtime graph only considers ipv4 127.0.0.1 for exclusion. If your system without dnsmasq is pure ipv6, then those addresses (like ::1) are not excluded. You also specified "127.0.0.0/8", but as you see below, luci only considers 127.0.0.1 as localhost.

----
The realtime conntrack data for the chart is coming via luci-bwc, and it has two alternative sources, 
"/proc/net/ip_conntrack" : "/proc/net/nf_conntrack"
https://github.com/openwrt/luci/blob/master/modules/luci-mod-admin-full/src/luci-bwc.c#L428
Does /proc/net/nf_conntrack exist in both of your scenarios?

And then the localhost connections are dropped if src and dst match:
https://github.com/openwrt/luci/blob/master/modules/luci-mod-admin-full/src/luci-bwc.c#L524
```
if (strstr(line, "src=127.0.0.1 ") &&
strstr(line, "dst=127.0.0.1 "))
```
It could be that the contents of /proc/net/nf_conntrack change when dnsmasq is installed. Is there a difference in the conntrack line format between your scenarios?

The realtime connection list gets its data via json (from luci-bwc, I think)
and again it is parsed for 127.0.0.1
https://github.com/openwrt/luci/blob/44bf3f0c1640040561306caff39f34ac916b4357/modules/luci-mod-admin-full/luasrc/view/admin_status/connections.htm#L140

EricLuehrsen, 2017-02-11:
It was obvious and you found it. Trying to keep track and I made a mistake. Its only IP6 and obscured by the name 'localhost'. Luck pull #1000.

EricLuehrsen, 2017-02-11:
It was obvious and you found it. Trying to keep track and I made a mistake. Its only IP6 and obscured by the name 'localhost'. Lucky pull #1000.

closed 2017-02-11 by EricLuehrsen