openwrt/luci #1006: luci-app-statistics broken links

koblack, 2017-02-12:
Firmware Version 
`LEDE Reboot SNAPSHOT r3418-b1c2506 / LuCI Master (git-17.042.68041-16395c3)`

Click on graph in Statistics/Wireless page and you get

`No page is registered at '/admin/statistics/graph/wlan0'.
If this url belongs to an extension, make sure it is properly installed.
If the extension was recently installed, try removing the /tmp/luci-indexcache file.`

The link is wrong
`/cgi-bin/luci/admin/statistics/graph/wlan0`

It should be?
`/cgi-bin/luci/admin/statistics/graph/iwinfo/wlan0`

Same on Processor/Processes page.

hnyman, 2017-02-12:
The same behaviour seem to be on all those statistics charts, where you have multiple data sources and separate tabs for them (e.g. wireless, processor, thermal, ...)

On the pages with only one data source, the charts are not links at all.

Possibly none of the charts should work as links.

koblack, 2017-02-12:
Yes, these links are not necessary.

hnyman, 2017-02-13:
Those links are created by /usr/lib/lua/luci/view/public_statistics/graph.htm

Originally the links have been added to the pages with multiple data sources by
https://github.com/openwrt/luci/commit/b2774f0cc64c3aec4328fe5869f410164a94cb89#diff-6c4c4de26cb3de6742abe230767ad23e

I patched my live router this way and the links disappear:
```
--- /usr/lib/lua/luci/view/public_statistics/graph.htm
+++ /usr/lib/lua/luci/view/public_statistics/graph.htm
@@ -29,9 +29,7 @@

 <div style="text-align: center">
        <% for i, img in ipairs(images) do %>
-               <% if is_index then %><a href="<%=pcdata(images[img])%>"><% end %>
                <img src="<%=REQUEST_URI%>?img=<%=img%>&#38;host=<%=current_host%>" />
-               <% if is_index then %></a><% end %>
                <br />
        <% end %>
 </div>
```

commit 298f838d (2017-02-13): luci-app-statistics: remove non-functional links from graphs

The statistics graphs on the pages with multiple data sources
have contained links to the data sources, but those links
have not worked. Remove the links as they are unnecessary.

This commit fixes #1006
closed 2017-02-13 by hnyman