openwrt/luci #1007: TL-WR842NDv1 LEDE 17.01.0-rc2: wrong switch port numbering

eripek, 2017-02-13:
On LuCI lede-17.01 branch (git-17.033.24085-e306ee6) I encounter the following issue:

cgi-bin/luci/admin/network/vlan:

The port status picture shown is '1'. The lan cable is plugged to the device plug number 4. Port LED 4 on the router is lit. The configuration of the port (untagged vlan 1 on port 4 and tagged vlan2 on 1,2,3) is correct for luci port 4, but not for port 1.

swconfig dev eth0 show:
The port numbers shown here match the lan led numbering of the router as well as the numbers on the yellow plastic port casing.

Furthermore it seems uci switch_port sections giving correct pvid options are missing.

hnyman, 2017-02-14:
Does nto sound like this possible bug is from LuCI. LuCI just reads the config data from /etc/config files.

Based on Openwrt wiki page for your device, the hardware manufacturer has connected the logical ports in a mixed way:
https://wiki.openwrt.org/toh/tp-link/tl-wr842nd#lan-port_mapping
```
chassis port 	internal-port
WAN (blue) 	CPU: eth0
1 (yellow) 	Switch eth1; Port 2
2 (yellow) 	Switch eth1; Port 3
3 (yellow) 	Switch eth1; Port 4
4 (yellow) 	Switch eth1; Port 1 
```
So yes, the external port 4 matches the internal/logical port 1 used in config.

> uci switch_port sections giving correct pvid options are missing.

Again, that has nothing to do with LuCI. The config itself is defined in the main LEDE & Openwrt code.

jow-, 2017-02-14:
@hnyman - sounds like it should get fixed in 02_network though

jow-, 2017-02-14:
@eripek - please provide the full output of `swconfig dev switch0 help`

eripek, 2017-02-14:
swconfig dev eth0 show:

```
Global attributes:
	enable_vlan: 1
Port 0:
	pvid: 0
	link: port:0 link:up speed:1000baseT full-duplex txflow rxflow 
Port 1:
	pvid: 0
	link: port:1 link:up speed:100baseT full-duplex auto
Port 2:
	pvid: 0
	link: port:2 link:down
Port 3:
	pvid: 1
	link: port:3 link:down
Port 4:
	pvid: 1
	link: port:4 link:down
VLAN 0:
	vid: 0
	ports: 0t 1t 2t 
VLAN 1:
	vid: 1
	ports: 0t 3t 4 
VLAN 2:
	vid: 2
	ports: 0t 2t 3t 
VLAN 3:
	vid: 3
	ports: 0t 1t 
```

swconfig dev switch0 help
```
switch0: eth0(AR7240/AR9330 built-in switch), ports: 5 (cpu @ 0), vlans: 16
     --switch
	Attribute 1 (int): enable_vlan (Enable VLAN mode)
	Attribute 2 (none): apply (Activate changes in the hardware)
	Attribute 3 (none): reset (Reset the switch)
     --vlan
	Attribute 1 (int): vid (VLAN ID)
	Attribute 2 (ports): ports (VLAN port mapping)
     --port
	Attribute 1 (int): pvid (Primary VLAN ID)
	Attribute 2 (unknown): link (Get port link information)

```

feckert, 2019-10-09:
We haven't heard back from you, so we're going to go ahead and close the issue.
If you're still experiencing the problem, please re-open the issue.

closed 2019-10-09 by feckert