openwrt/luci #1013: last commit https://github.com/openwrt/luci/commit/7e6b3d1eb93aac0ae64468abbfb05cc4a46652ed luci-mod-admin-full = not possible configure any other config wireless or modification default (LEDE-17.01-SNAPSHOT) from Firefox to ramips/WITI

build000, 2017-02-15:
as the topic

hnyman, 2017-02-15:
What exactly is the problem? Which error you get? And what you are doing when you get it?

There was a HTML problem for "openwrt" theme, but the code should have worked ok with "bootstrap" or "material" theme. I fixed that a few minutes ago with https://github.com/openwrt/luci/commit/a76564d9735082a7b42f5fa6d46f60791228f5a8  
(currently compiling my own 17.01 build to verify)

build000, 2017-02-15:
yes, this error if use "openwrt theme" - just pops up an error page with debugging firefox and nothing can be done with the configuration of the luci - then only helps the modification of the shell via ssh.

hnyman, 2017-02-15:
> this error if use "openwrt theme" 

That should be now fixed with https://github.com/openwrt/luci/commit/a76564d9735082a7b42f5fa6d46f60791228f5a8, like I said earlier.

I just tested with LEDE Reboot 17.01-SNAPSHOT r3150-367a3bb36f / LuCI lede-17.01 branch (git-17.046.24899-a76564d)  and all three themes (Bootstrap, Material, Openwrt) work ok with both wpad-mini and wpad.

build000, 2017-02-15:
ok, thx :D

hnyman, 2017-02-15:
If you want to quickly test the fix without recompiling, just manually edit the changes from https://github.com/openwrt/luci/commit/a76564d9735082a7b42f5fa6d46f60791228f5a8 
into /usr/lib/lua/luci/model/cbi/admin_network/wifi.lua in the live router

There are five places with `<br>` that should be `<br />`. That is the only change.  (strange that Openwrt there chokes on that, but others work)

build000, 2017-02-15:
I confirm - described by you fast file modification /usr/lib/lua/luci/model/cbi/admin_network/wifi.lua brings the expected recovery of the primary funkcjaonalności (device settings wifi).
-once again thank you for your prompt response. :D

closed 2017-02-15 by build000

hnyman, 2017-02-15:
@jow- 

>  There are five places with `<br>` that should be `<br />`. That is the only change. (strange that Openwrt theme chokes on that, but other themes work)

I think that I figured that out. For some reason the old "Openwrt" and "freifunk-generic" themes enforce "XHTML 1.0 Strict" formatting standard, while the Bootstrap and Material themes do not. And xhtml-strict requires all html elements to be "closed" (with /).

That dates to 2008: https://github.com/openwrt/luci/commit/77b05af325beca587040e1a429e2ad20d2849b7a
"Add stupid XHTML 1.0 DOCTYPES to themes" 

https://github.com/openwrt/luci/blob/master/themes/luci-theme-openwrt/luasrc/view/themes/openwrt.org/header.htm#L123
`<!DOCTYPE html PUBLIC "-//W3C//DTD XHTML 1.0 Strict//EN" "http://www.w3.org/TR/xhtml1/DTD/xhtml1-strict.dtd">`
I am not quite sure that it is necessary to enforce that on just one theme. Maybe the restriction should be removed?

build000, 2017-02-15:
I think it's not a good idea - is about to spill half functionality luci on the theme "openwrt"...