openwrt/luci #1021: luci-app-statistics processes missing graphs

koblack, 2017-02-15:
Firmware Version 
`LEDE Reboot SNAPSHOT r3498-4037445 / LuCI Master (git-17.046.32957-dea880e)`

Statistics/Processes does not show all graphs also the legend looks strange.

hnyman, 2017-02-15:
When has it worked for you last time? 
(is this a recent change?)

koblack, 2017-02-15:
That occusrs on CC / BB / AA I don't find any working version

hnyman, 2017-02-16:
I installed the processes module to LEDE 17.01 and the graphs seem to work ok. The top graph is only shown in the overview page

[image]

koblack, 2017-02-16:
Yes, on the overview page is missing few graphs and also specific process subpage e.g uhttpd missing one top graph. Is this the expected behavior?

[image]

hnyman, 2017-02-16:
It is at least partially expected, as there is just one common page layout for the overview page and the process-specific pages.

the "top chart" = system level data that exists only on system level, so it will not be visible on process-specific subpages. But it has a placeholder also on them.

I noticed that same regarding the overview page that only the cpu time is showed for each process on the overview page. It is not by design, but most likely due to the data structure. Might be that it could be fixed with some editing, but all charts are visible on some page, so this problem is not that big.

(I though originally based on your first message, that there are no charts at all.)

koblack, 2017-02-17:
file `/usr/lib/lua/luci/statistics/rrdtool/definitions/processes.lua`

The difference between cpu time and others is "detail = true". Removing that option cause process-specific graphs is also visible on overview page. Maybe is good idea to remove it completely or add also to cpu time definition? The page will be more readable.

Goro2030, 2017-04-29:
@koblack , any news on this? I'm facing the same issue... this is what appears in the syslog when I try to see the processes tab:

```
Sat Apr 29 17:45:32 2017 daemon.err uhttpd[2559]: ERROR: opening '/mnt/sda1/RouterACS/processes/ps_cputime.rrd': No such file or directory
Sat Apr 29 17:45:32 2017 daemon.err uhttpd[2559]: ERROR: opening '/mnt/sda1/RouterACS/processes-bandwidthd/ps_state-sleeping.rrd': No such file or directory
Sat Apr 29 17:45:32 2017 daemon.err uhttpd[2559]: ERROR: opening '/mnt/sda1/RouterACS/processes-collectd/ps_state-sleeping.rrd': No such file or directory
Sat Apr 29 17:45:32 2017 daemon.err uhttpd[2559]: ERROR: opening '/mnt/sda1/RouterACS/processes-darkstat/ps_state-sleeping.rrd': No such file or directory
Sat Apr 29 17:45:32 2017 daemon.err uhttpd[2559]: ERROR: opening '/mnt/sda1/RouterACS/processes-dnsmasq/ps_state-sleeping.rrd': No such file or directory
Sat Apr 29 17:45:33 2017 daemon.err uhttpd[2559]: ERROR: opening '/mnt/sda1/RouterACS/processes-dropbear/ps_state-sleeping.rrd': No such file or directory
Sat Apr 29 17:45:33 2017 daemon.err uhttpd[2559]: ERROR: opening '/mnt/sda1/RouterACS/processes-hostapd/ps_state-sleeping.rrd': No such file or directory
Sat Apr 29 17:45:33 2017 daemon.err uhttpd[2559]: ERROR: opening '/mnt/sda1/RouterACS/processes-smbd/ps_state-sleeping.rrd': No such file or directory
Sat Apr 29 17:45:33 2017 daemon.err uhttpd[2559]: ERROR: opening '/mnt/sda1/RouterACS/processes-snmpd/ps_state-sleeping.rrd': No such file or directory
Sat Apr 29 17:45:33 2017 daemon.err uhttpd[2559]: ERROR: opening '/mnt/sda1/RouterACS/processes-uhttpd/ps_state-sleeping.rrd': No such file or directory

```

Any tips?

hnyman, 2017-04-30:
Like I explained above, it is mainly due to the simplistic structure of the statistics graph pages. The same page is serving both the "global overview" and the process-specific detail pages (selectable as tabs from the top menu). Not all data is available in both the global and the detail pages, and that makes some graphs to fail. See the difference in data contents of global & process-specific dirs:

```
root@LEDE:/tmp/rrd/LEDE# ls processes*
processes:
fork_rate.rrd          ps_state-paging.rrd    ps_state-sleeping.rrd  ps_state-zombies.rrd
ps_state-blocked.rrd   ps_state-running.rrd   ps_state-stopped.rrd

processes-dnsmasq:
ps_code.rrd        ps_cputime.rrd     ps_pagefaults.rrd  ps_stacksize.rrd
ps_count.rrd       ps_data.rrd        ps_rss.rrd         ps_vm.rrd

processes-dropbear:
ps_code.rrd        ps_cputime.rrd     ps_pagefaults.rrd  ps_stacksize.rrd
ps_count.rrd       ps_data.rrd        ps_rss.rrd         ps_vm.rrd

processes-uhttpd:
ps_code.rrd        ps_cputime.rrd     ps_pagefaults.rrd  ps_stacksize.rrd
ps_count.rrd       ps_data.rrd        ps_rss.rrd         ps_vm.rrd
```

The page would need much more complex logic to clean this up.

And as the count of monitored processes can be rather large, I see no point in cramming all charts to the main page.

koblack, 2017-05-08:
> When has it worked for you last time?
> (is this a recent change?)

last working version
http://luci.subsignal.org/trac/browser/luci/trunk/applications/luci-statistics/luasrc/statistics/rrdtool/definitions/processes.lua?rev=3591

@hnyman @Goro2030 look at this file (added labels etc.)
https://gist.github.com/koblack/1b5b83aab2f5548b0a585ca0e4ee51c3

After change everything works fine.

hnyman, 2017-05-08:
What about that file?
That seems to implement some of the improvements that I did 8 days ago with https://github.com/openwrt/luci/commit/b737831e56854f55fedbc6a6394d385ee0aef6e3 (and backported to LEDE 17.01 with 9686fd1cc40e49799443cbd3eea832ae2a0cc07c )

hnyman, 2017-05-08:
hmm. I now noticed that you also added the plugin_instance detection.

So you removed the CPU time charts from the main top page, and the summary chart from the details pages, right?

hnyman, 2017-05-08:
I tested your detection change on a live router running LEDE master and the change works.
(Seems that the change has no impact on the visibility of the CPU time chart on the main page. I got that wrong in my my previous message.)  

Nice.
Interested in submitting a pull request? 
(I can also do the necessary change myself.)

Goro2030, 2017-05-08:
It works GREAT now @koblack !! Thanks for doing this .... you should make the change part of the trunk asap.

hnyman, 2017-05-09:
Fixed by https://github.com/openwrt/luci/commit/0a6ce204234c9485549e290230e51484b1fceb64

closed 2017-05-09 by hnyman